SeedSigner/seedsigner #252: Allow Screen Brightness Adjustment On SeedQR Screens

BitcoinQnA, 2022-09-12:
Is it possible to allow for screen brightness adjustment on these screens in the exact same way you already offer with XPUBs and PSBTs?

SeedSigner, 2022-09-18:
I believe @kdmukai has already privately remarked that this was a great catch and should absolutely be implemented. We should leave this open to make sure it stays on the radar.

SeedSigner, 2025-12-18:
As I think more on this, have to agree with @kdmukai here. Even when brightness is appropriate, it can be really difficult to scan the QR rendered at this resolution, and as Keith alludes, we don't want to normalize the scanning of pure seed / key material from the screen without carefully thinking about a specific use case, and also thinking through the ramifications of that. Given the preceding, and the age of this issue, going to close.