SeedSigner/seedsigner #258: Feature Request: UI Change Dice Number input

smartsourcefreak, 2022-09-25:
Subject: Change UI for dice number input to make it way faster and the control less complicated.

Suggestion: Use the joystick (UP=1, RIGHT=2, DOWN=3, LEFT=4) and the UPPER two buttons on the right (TOP=5, MIDDLE=6) to let the user enter the dice numbers without having to extra confirm each number. Assign the 3rd button on the right (BOTTOM=BACKSPACE) to let the user delete the last entry. Show graphically on screen all numbers and how they are assigned to the joystick and the buttons, similar to the I/O Testscreen.

Once the entropy is completely put in, move to the usual "Caution / "I understand" screen.

SeedSigner, 2022-09-28:
This is a creative solution to make inputing dice rolls to go much more quickly. We would also probably need to add a notification screen before the input process began to explain to users which controls correlated with which dice rolls. I would be curious to get opinions on this approach from Nick or Keith.

smartsourcefreak, 2022-09-28:
> We would also probably need to add a notification screen before the input process began to explain to users which controls correlated with which dice rolls.

Not necessary if it's done like I said above: While in "dice roll input mode" show a graphic exactly like the I/O-test-screen. Similar to the graphic I attached. No further explanation needed this way.

[image]

SeedSigner, 2022-09-30:
Another thought I have is to utilize the center press for 5, which frees up a button on the right, that could be used for an "abort" option to allow a user to back out without having to finish the entire entry process.

smartsourcefreak, 2022-09-30:
I thought about it as well. But I came to a different conclusion: Pressing the joystick button is the least comfortable button push and if you slide to the side by not pressing straight down you end up with a different key input. But the 5 will be demanded to be pushed way more often than "ABORTing" the process. So I'd assign the "ABORT-feature" to the joystick button push and remain the buttons to the right as suggested. On top if you press the joystick button, instead of instantly aborting the dice rolling process, the user could be requested to either "ABORT" the process, "CONTINUE" the dice rolling. Maybe even "CREATE SEED NOW (insecure)".