SeedSigner/seedsigner #259: shasum -a 256 -c seedsigner_0_*_*.img.zip.sha256 throws a warning

morriswinkler, 2022-09-27:
on OSX the shasum verification throws a warining:

```
█ shasum -a 256 -c seedsigner_0_*_*.img.zip.sha256
seedsigner_0_5_1.img.zip: OK
shasum: WARNING: 4 lines are improperly formatted
```

It is caused by the PGP header and footer in the sha256 file
```
-----BEGIN PGP SIGNED MESSAGE-----

bc04eb573bdb7cac3d7ab9b1b44850ab704a8dad1041a52c4527fe6aa5b8f33c  seedsigner_0_5_1.img.zip

-----END PGP SIGNED MESSAGE-----
```

The shasum command line tool doesn't understand line 1/2/4/5.

I guess that is not an issue if you know what is going on, but if you do not you might suspect that the binary images was tempered with.

SeedSigner, 2022-09-28:
The warning about the improperly formatted lines can be disregarded, what matches is that the sha256 hash of the .zip file matches what was published with the release. Kudos to you for going through the verification process! I would like to have as many people doing this as we can have.

morriswinkler, 2022-09-28:
As I sad, I totally understand what the warnings mean, but I am just imaging someone that uses a command line the first time might be scared off by the warning.

newtonick, 2022-11-29:
@SeedSigner I agree if the [seedsigner_0_5_1.img.zip.sha256](https://github.com/SeedSigner/seedsigner/releases/download/0.5.1/seedsigner_0_5_1.img.zip.sha256) file matched the same format that Sparrow Wallet uses (as an example, ie [sparrow-1.7.1-manifest.txt](https://github.com/sparrowwallet/sparrow/releases/download/1.7.1/sparrow-1.7.1-manifest.txt)) this would remove this issue.

SeedSigner, 2022-11-30:
Appreciate you weighing in on this, I will try it with the next release -- could make sense as I think on it to simplify things b/c we will now have multiple binaries with the parallel Raspbian/Buildroot releases.

Marc-Gee, 2023-02-28:
This has been fixed in the 0.6.0 manifest (.sha256) file. ie There are no longer any extra lines in that [Manifest] file. 
Hence the shasum command no long returns the warning message:  "shasum: WARNING: 4 lines are improperly formatted"