SeedSigner/seedsigner #260: Verify Address Issue With Nunchuk On Testnet Only

BitcoinAscent, 2022-10-05:
When I use Seed Signer with Nunchuck on mainnet, everything works great!

But when I switch to testnet, export a xpub into Nunchuk, create a wallet with that xpub, I can't verify an address. The Seed Signer just keeps looking for that address counting up into the thousands of addresses until I tell it to stop looking.

This verify feature works just fine for me when Nunchuk is connect to a mainnet wallet. The issue is only on testnet. The Seed Sign knows it's looking for a Native Segwit Testnet Address, it just can't find it.

The fingerprint and derivation path on the Seed Signer and Nunchuk all match up, so I don't understand why the Seed Signer can't find and verify the address.

My Seed Signer version is 0.5.1

[image]

kdmukai, 2022-10-06:
This looks like the source of the bug:
https://github.com/SeedSigner/seedsigner/blob/dev/src/seedsigner/views/seed_views.py#L1545

It's grabbing the `network` from `Settings` instead of the value that was determined when the address was scanned. In other words: the SeedSigner should be able to be configured for `mainnet` in `Settings` but still be able to verify a `testnet` address. But in this case it's incorrectly passing `mainnet` into the xpub generation.

@BitcoinAscent, can you try changing the network in `Settings` -> `Advanced` to `testnet` and trying again? This bug should be fixed but at least we'd have verification that my reading of this is correct.

BitcoinAscent, 2022-10-06:
Yes... I tried that before and just tried it again. Seed Signer still can't find the address. I also changed the network to "regtest". It couldn't find it there either.

Just let me know if there anything else you'd like me to try.

newtonick, 2022-11-29:
@BitcoinAscent If you have testnet seed words and a address, I can test to see if I can reproduce. Otherwise I'm not sure how to help resolve this for you.

newtonick, 2022-11-29:
@BitcoinAscent I create a key/wallet in nunchuck using the same seed in both testnet and mainnet. I was able able to verify the first address in both without issue. I'm guessing you used the same xpub from mainnet in testnet (which you can't do because the derivation paths are different).

I asked for your specific seed phrase to confirm the address you were testing the screenshot above is from the wrong network xpub.

BitcoinAscent, 2022-11-29:
@newtonick 
Oh, ok.
But as you can see from my picture above, the SeedSigner knows it's looking for an address on the testnet.

Do I need to switch the Seedsigner to testnet BEFORE I export the xpub to Nunchuk connected to testnet?

newtonick, 2022-11-29:
> Do I need to switch the Seedsigner to testnet BEFORE I export the xpub to Nunchuk connected to testnet?

correct, because native segwit bip84 has different standard derivation paths for different networks.

closed 2022-11-29 by newtonick