SeedSigner/seedsigner #269: Feature Request: Border Wallets integration

SuperPhatArrow, 2022-11-02:
[What if SeedSigner & Border Wallet had a baby?](https://stacker.news/items/56513) by HardRich

If you are not familiar with Border Wallets you can find out more about it at [borderwallets.com](https://www.borderwallets.com/)

Over the last few months, @SeedSigner @microchad @kdmukai and I have been discussing how to do this and here is a summary of our discussions so far...

The idea here is to allow SS users to be able to input their seed with a pattern as opposed to 12/24 words. They can either input their grid regeneration phrase or scan a GridQR and the Seed Signer device will regenerate the grid for them to input their pattern.

Information on how the deterministic grids are created can be found [here](https://github.com/microchad/borderwallets/blob/gh-pages/TECHNICAL.md) and a python lib for the UHERNG can be found [here](https://github.com/wuftymerguftyguff/uheprng)

GridQR is the QR Code we decided on for sharing Deterministic Entropy Grids. It is formatted as `bweg:<URI encoded grid regeneration phrase>?v=<version number>` Examples:

[image]
`bweg:industry%20palace%20melt%20still%20wreck%20unfair%20ivory%20absorb%20brother%20few%20rigid%20toy?v=1`

[image]
`bweg:border%20wallets%20and%20seed%20signer%20had%20a%20baby?v=1`

GridQR will be included in a future release of border wallets.

The next issue was fitting a 16x128 grid onto a 240x240 screen and @easyuxd  came of with the Multilevel zoom:

[image]

I have been asked by the group to submit a summary as an FR issue so we can open this up to comments and have a reminder of all the details.

microchad, 2022-11-24:
Very excited about this. Some things to consider:

1) The navigation should maintain consistency with Column and Row labelling when zooming in - i.e. reflecting which row and columns are depicted at each zoom level (A-P, 001-128) and marked as such
2) Final word needs to be inputted or selected by the user after inputting their Pattern
3) Support for Final Word Number feature - where user is asked to input a number, which corresponds to their final word

dadofsambonzuki, 2024-05-21:
Should this also include Pattern Grid generation and the random production of the Entropy Grid using the same methods used for private key generation in SS?

microchad, 2024-05-21:
> Should this also include Pattern Grid generation and the random production of the Entropy Grid using the same methods used for private key generation in SS?

Yes. Allowing users to create a Border Wallet using SeedSigner, and not just recreate it, would be ideal.