SeedSigner/seedsigner #273: Prudently Paranoid Settings

jdlcdl, 2022-11-23:
...for the prudently paranoid user.

Use case:
* User has disabled: Xpub export, Show Xpub Details, and/or a new setting "Seed backup" (which removes this option from the seeds menu)... because these steps have already been done long ago.
* Now the user is moving about, with seedsigner on a battery pack, collecting seeds without turning it off, and continuing on to load more seeds.
* The user's threat model is rapidly experienced increased susceptibility to a violent attack as they load more seeds.

As a solution: Some security-sensitive settings could raise an exception whenever these settings are toggled from disabled to enabled, alerting the user with "You must discard all seeds before enabling this option!"

Thoughts?

jdlcdl, 2022-11-29:
After sleeping on this, I've gone on to expand what is covered by Prudently Paranoid Settings, including sig-types and script-types, but the initial idea hasn't changed.  It's the seedsigner assumption, always erring on the side of caution, that "disabled means disabled and I won't relax these privacy/security settings while keys are loaded."

I've implemented this using existing/new settings definitions, and a new method "settings.is_enabled(a_setting)" that can be used like a question (returning True or False) or like an assertion (raising an exception if disabled).  Within low-level views that arguably should not be accessible when a particular setting is disabled, they'll use settings.is_enabled(a_setting, assert_=True) to raise an ugly exception, disrupting continued flow.  To spare the user from seeing these, higher level views will ask "is_enabled(a_setting)" to avoid offering navigation to functionality that is disabled.

revetuzo, 2022-12-15:
Can you explain the use-case of traveling around and collecting seeds? To sign a transaction, you can just use partially signed bitcoin transactions (PSBT), one seed after another.

jdlcdl, 2022-12-15:
Key material is split in pieces (seedxor style), physically separated (perhaps on the same property, perhaps a short walk, maybe a short drive).

This use case starts at loading seed #0, from the moment you load it into seedsigner and could have had a chance to secure it by putting your seed-qr back in the safe and locking it, to the point you shutdown/pull-power.  With prudently_paranoid_settings: you might have seed-backup, address-explorer, and export-xpub disabled... meaning these tools would not leak their info... and seedsigner would still be capable to sign a PSBT from a known wallet... less capable to attacker who doesn't already know your utxos.

jdlcdl, 2022-12-15:
I also like what it does to the assumption of:

If you see someone with a seedsigner who just had it turned on... they've got full knowledge, or recorded seed material on them at this very moment.

That would not be a correct assumption if you and I were face to face and I just signed a transaction in front of you.  A better assumption would be that I'm not even capable of loading my keys without running around in front of public cameras... and even if you can take my seedsigner from me while it's still turned on... you won't learn more than my fingerprint.

jdlcdl, 2022-12-16:
> Can you explain the use-case of traveling around and collecting seeds? To sign a transaction, you can just use partially signed bitcoin transactions (PSBT), one seed after another.

I am not building an argument that scanning a psbt 2x per seed, with distance/obstacles/people in between is to be replaced by prudently_paranoid_settings.

I also will not rebut the argument that it could imprudently enable users to take bad habits, where keys that "ideally" should be separated end up hanging-out together regularly.  Realistically, I suspect this already happens in private households, that multiple keys are stashed discretely yet accessible, and I'm arguing that prudently_paranoid_settings might be desirable in the majority of seconds between scanning those accessible yet hidden/locked-up SeedQRs and locking them back up immediately, signing whatever is to be signed, and powering off.

jdlcdl, 2022-12-16:
As far as sig_type and script_type being included... which add a little bit of clumsy granularity, this was just me fleshing out the idea to its extremes... that are likely not so useful.   It'd be a stretch to imagine the usefulness of a completely disabled seedsigner.   Anyways, for me to propose ideas as complete as I can + an implementation, and for others to pick and choose.

kdmukai, 2022-12-16:
Concept NACK.

> [added]: Consider the amount of seconds in the lifetime that seedsigner is running with any particular seed loaded. 

This is the most compelling part of the argument. But I keep coming back to two basic counterpoints:

1. I don't like the idea of relying on UI settings as a form of security. We do allow for an increasingly flexible way of turning features on or off in Settings, but that's meant to alter the user flows to customize it for experts vs noobs. Using the UI as a security barrier does provide some value but isn't strong enough to live up to more naive users' expectations. Even with all the appropriate warnings, I'm still hesitant to encourage anything that could create a false sense of security. Plus if the most likely use case is during the short time the seed is loaded, then the seed backup words/SeedQR would be in the vicinity and would remain the bigger risk vector.

> but if among all that time you get distracted... 

2. I've thought for a while that we should have an optional "Wipe after X time" Setting. Right now a SeedSigner will hold any seeds in memory for as long as it sits plugged in. A configurable option to wipe after 5 min, 10 min, X hrs, etc of idle time would be a nice safety measure that should probably default to ON. And/or a secondary setting that powers down the SeedSigner after X time.

EverydayBitcoiner, 2022-12-30:
> 2. I've thought for a while that we should have an optional "Wipe after X time" Setting. Right now a SeedSigner will hold any seeds in memory for as long as it sits plugged in. A configurable option [...]

This was discussed in issue #142.