SeedSigner/seedsigner #279: QR abliities for Xpub

alexjrlewis, 2022-12-06:
Add the ability of an Xpub to be exported to a QR code in the same way as a 12-/24-word seed can be.

Update paper templates to include an Xpub version. This will be useful for importing public keys ala SeedHammer method.

jdlcdl, 2022-12-08:
I believe there are already multiple formats to export an Xpub from seedsigner, but you might take a look at the BlueWallet format.  For me, It's a single image (instead of animated qr), encoding the xpub like: "[fingerprint/bip/32/hd/path]base58-check-encoded-Bip32-extended-public-key".  This would be a tedious challenge to transcribe by hand, but might photograph it (polaroid for ultra-privacy-paranoid mode?) since it's public?

Or, might a solution already exist outside of seedsigner.  Perhaps your watch-only wallet can give you what you want, like a wallet descriptor which is a superset of the xpub, and will include all you need for multisig (script-types, master xpubs, paths, etc).  On Specter, I get a single qrcode that would be challenging to transcribe by hand, but could be photographed or sent to a device that's going to print/scribe it permanently.

I'm assuming that the use case is to have additional qr's that seedsigner can quickly read... which you trust because they were setup in advance and stored securely/not-modified?  Or maybe this is for ease of using different wallets?

If I've completely misunderstood your post, and/or the use case, please forgive me (my excuse: im new here).

al21e6, 2022-12-08:
> I believe there are already multiple formats to export an Xpub from seedsigner, but you might take a look at the BlueWallet format. For me, It's a single image (instead of animated qr), encoding the [...]

More for saving multisig in xpubs such that any 2 seed + alternating xpubs can load a 2 of 3 multisig wallet:

https://twitter.com/SeedHammer/status/1598247830678048768

So if we could have similar qr templates and use the "export qr code to paper" tool provided for 12/24 word seeds (i.e. where we manually copy the seed words to a qr template ourselves)

SeedSigner, 2022-12-12:
Hmmmm... Not sure if manually transcribing XPUB would be a reasonable process or not as they typically contain significantly more data than a seed phrase. Another thing to keep in mind is that the XPUB, is private, but not secret, information, so you may feel comfortable taking a photograph of a single-frame XPUB, or videoing an XPUB QR set to document each frame?

EverydayBitcoiner, 2022-12-20:
So I made a quick python script based off of @SeedSigner's baseball cards and SeedHammer's backup scheme that can be found here: https://github.com/EverydayBitcoiner/multisig-backup. It needs a little polishing, but works.

EverydayBitcoiner, 2022-12-20:
The only way that I see this integrating directly into the seedsigner device would be to have the seedsigner generate a pdf document that could be transferred through qr to the computer for printing, but since the descriptor (i.e. all the xpubs) is already on the coordinating device (desktop, phone) then is seems like it would be ok to use that device to generate the multisig backup like I did with my code. Seems unnecessary IMO to integrate this into the seedsigner device software.

EverydayBitcoiner, 2022-12-27:
> So if we could have similar qr templates and use the "export qr code to paper" tool provided for 12/24 word seeds (i.e. where we manually copy the seed words to a qr template ourselves)

NACK

This seems like an excessive step that would fall outside normal use. As @SeedSigner noted above XPUBS do not require the same level of secrecy as seeds and the size of an xpub qrcode is significantly larger than that for a seed (looks like seedsigner BW format is ~42x42 which is over twice as many dots to fill in as the largets 29x29 seedqr).

Unless we can find a compelling scenario for this use case that can't be better achieved with other tools then I am leaning against implementing something like this.

OliverOffing, 2023-05-08:
We could mitigate eventual loss of funds by being extra clear to the user that in a multisig setup—even if you set it up so that only n of m signatures are needed to move funds—you will need all of the xpub keys in order to retrieve the funds.

I suggest adding an extra screen:
1. User clicks "Export Xpub"
2. User clicks "Multisig"
3. User is shown "Privacy Leak" screen
4. **(New screen)** User is shown "Make sure you backup all xpubs for your multisig" screen
5. User is shown the xpub QR code.

With regards to storing this xpub key digitally, I'm not super happy about it. The whole point of having a physical seed that gets read on-demand using a stateless device is that you won't need to store anything digitally to recover funds.

The Polaroid idea is nice—if you have one. I'd be interested in hearing about other possible solutions beside "scan it digitally and print it."

---

> It is absolutely crucial that each cosigner’s backup includes the [xpubs] of all other cosigners […]. ([source](https://medium.com/shiftcrypto/the-pitfalls-of-multisig-when-using-hardware-wallets-9b0e98e4c19c))