SeedSigner/seedsigner #287: [Enhancement] Long Click Functionality

EverydayBitcoiner, 2022-12-24:
Implementing functionality to enable use of long click for any buttons. This came up in https://github.com/SeedSigner/seedsigner/pull/286#issuecomment-1364563189 where the use case would be for long pressing the back button to abort all flows and go all the way back home instead of having to go back through the whole backstack.

Another possible use case would be to avoid accidental clicks for certain actions such as you have to long press the discard seed button when getting rid of a seed (I know that there is already a screen warning before this step). I think this could be a good functionality to have as it opens up different ways of developing the UX.

EverydayBitcoiner, 2022-12-24:
On the topic of long press to go home, maybe it could be used so that if you held the joystick left (long press) it would take you home from wherever you are.

kdmukai, 2022-12-26:
Long-press on click is a well-established (albeit sometimes overly invisible) UX. But holding LEFT is used on keyboard screens to auto-advance across the keys. Even on non-keyboard screens, I think the expectation is that holding LEFT would just pin you to the side. We deliberately opted to not try to mimic a page swipe UI, so there's (almost) no left/right navigation in the UI (image entropy is an exception but probably could use a UI re-skin).

Either way, we definitely want to get @easyuxd to weigh in on long-press.

easyuxd, 2022-12-31:
Thanks for the navigation enhancement proposals @EverydayBitcoiner!

Back navigation could be a good use case for the long press. It would need to trigger a dialog, though, to avoid accidentally losing all progress in a flow. I’m thinking of all the different enclosures and 'stabby nub' alternatives that are manifesting in the community that may impact things like long press.

What do you think of adding a dedicated Cancel button to the navbar instead? It would provide a clearer visual affordance while avoiding unintentional long-press interactions. @kdmukai, you may recall an earlier design similar to this:

[image]

——

Regarding deleting a seed (or another major/destructive action) I think the warning dialog is enough, and best practice. Unless the long press is being used ubiquitously across the UI, I’m concerned that the interaction will be opaque to users. It might be made more intuitive for use in CTAs by adding additional content (“Hold to delete”) and then providing a progress bar for the hold inside the button. But off the top of my head, I can’t think of a use case where this would work better than a simple dialog screen.

EverydayBitcoiner, 2023-01-01:
> What do you think of adding a dedicated Cancel button to the navbar instead? It would provide a clearer visual affordance while avoiding unintentional long-press interactions. @kdmukai, you may recall an earlier design similar to this:

I think something like this is very intuitive for most users. The main obstacle is the fact that it takes up some of the title bar space which could make the naming of screens a little more challenging. If we went with this route I think that the X button could also be used as a home button when you are on a screen that doesn't necessarily have a flow or have any pending information, but is more than one click to go back to the main menu, such as something in the advanced menu setting.

SeedSigner, 2023-01-02:
This is good discussion. For me there's some cognitive dissonance with the exit being on the right side of the screen, as it is the ultimate "back" and I associate back with the left side of the screen. (Though I of course realize this isn't a formal proposal or anything, it's just an observation on my part). It can be cumbersome when you just want to exit an entire workflow and have to use the back button multiple times, so I definitely understand the use case for this and have been there myself as a user. It brings up the question of how much we can/should communicate through cues in the UI, and what things may just be advanced UI features that users might stumble on themselves or read about / hear about from someone else. But then again, is IYKYK good practice in UI? Advanced PC users know all kinds of keyboard shortcuts that are invisible to less advanced users. Maybe a long press on the back arrow could activate a progress animation that once complete would trigger the discard-progress notification and take you all the way back to the main menu? Just throwing out some ideas, I do find this discussion useful.

EverydayBitcoiner, 2023-01-02:
>For me there's some cognitive dissonance with the exit being on the right side of the screen, as it is the ultimate "back" and I associate back with the left side of the screen.

Interesting. The X in the top right makes complete sense in that it is almost universally used throughout UIs to indicate closing out or exiting something. Even closing tabs in Android browser users the X in the top right. As far as the UI goes I think that it is intuitive and easily understandable (maybe could even put the X in red for extra cues that you are doing something that can't be undone). I think that when you are not in a flow (like when you are deep in settings where there is nothing to save it loose) instead of an X you can just have a little home icon.

SeedSigner, 2023-01-03:
Maybe my problem is that I'm a mac user.  🤪

Cognitively I always kind of "imagine" screens moving from a left to right sort of progression, but maybe I'm an anomaly in that?

[image]

EverydayBitcoiner, 2023-01-04:
😂 it's been so long since I've used a Mac, I actually thought about it and thought that it was the same as Windows and Linux (at least the flavors I've tried). Should've verified 🤦‍♂️.

Here we are where the Mac vs. PC scenario again 🤪. I use mostly Linux and some Windows so I'm very used to the X in the upper right and hence it is very intuitive to me.

kdmukai, 2023-01-04:
> What do you think of adding a dedicated Cancel button to the navbar instead? It would provide a clearer visual affordance while avoiding unintentional long-press interactions. 

An additional icon in the TopNav title bar is a no-go. We're already too constricted for space up there. If we took any more space away, the German translation would have to just be one single screen at launch that says, "Sorry, learn English" (but in German, of course!).

EverydayBitcoiner, 2023-01-04:
> An additional icon in the TopNav title bar is a no-go. We're already too constricted for space up there. If we took any more space away, the German translation would have to just be one single screen at launch that says, "Sorry, learn English" (but in German, of course!).

That was main concern with an additional button on the navbar given that even in English it would crowd the space. I didn't think about it being even more of an issue in a different language.

SeedSigner, 2023-01-06:
Does anyone have any thoughts on long press triggering a transition animation of the "<" to an "X"? If the animation is permitted to complete, then the user gets a pop-up about exiting the workflow with info discarded. Wouldn't confuse or get in the way for less experienced users, but available for power users when they need it?

EverydayBitcoiner, 2023-01-07:
> An additional icon in the TopNav title bar is a no-go. We're already too constricted for space up there. If we took any more space away, the German translation would have to just be one single screen at launch that says, "Sorry, learn English" (but in German, of course!).

As an out of the box idea how about a horizontal scrolling title?

[ < ]  Address Exp...  [ x ]
[ < ]  ddress Expl...   [ x ]
[ < ]  dress Explo...   [ x ]
[ < ]  ress Explor...    [ x ]
[ < ]  ess Explore...   [ x ]
[ < ]  ss Explorer       [ x ]

EverydayBitcoiner, 2023-01-15:
Like this https://twitter.com/KeithMukai/status/1614747103020060672?s=20&t=OQh2y-Z67IXrNaoAK5YR4A.

@kdmukai is always a step ahead.