SeedSigner/seedsigner #288: [Feature] Warning on shutdown

EverydayBitcoiner, 2022-12-25:
As I've been poking around on seed signer and playing around with some other HWW/signing devices I am wondering if there should be a warning when a user goes to shutdown or restart making sure that the user knows that the seeds will all be forgotten on power off. This is something that is obvious to power users, but I'm thinking about the newbiests of newbs and making sure they don't misunderstand and think their keys will be stored over restart because a different wallet they have used did that. As I'm thinking about this I also am conscience that we do not want to overuse warning pages, but thinking this could be a possibility.

I am not sure how this would interact with the (I think reasonable) feature request for auto shutdown or lock after given period of time in issue https://github.com/SeedSigner/seedsigner/issues/142.

jdlcdl, 2022-12-25:
This is just my very personal opinion on noobs vs powerusers, I'll never be the one to step much out of my way to protect noobs from themselves while pushing the cost/noise/confusion/even-an-extra click on to powerusers.  The only case where this warning might be helpful to a noob is when seedsigner has created a brand new seed since it was powered on, and at that time the user has already seen warnings, reviewed seedwords and verified/skipped-verification; all other uses of seedsigner imply that the user has safely backed up their seed or decided to commit it to memory.  Perhaps if a noob creates a seed, exports xpub and creates a wallet in coordinator, and funds utxos all in the same session -- without backing up their seed -- they deserve to learn the hard way?

There is some free screen real-estate in the power options menu where perhaps a warning that the next action will nuke machine state... and I happen to like it clean, but a warning triangle might be appropriate there.

This might be a strong non-friendly noob opinion, but it's just mine... it stems from my very strong belief that everyone is capable and that sloppy incompetence is mostly a lack of will and rarely a lack of capacity.

kdmukai, 2022-12-25:
> all other uses of seedsigner imply that the user has safely backed up their seed or decided to commit it to memory.

This is a well-argued point. I mostly agree with @jdlcdl.

That being said, rather than have a warning, I could see some value in simply stating the fact: "All seeds will be cleared from memory" or something like that. It's not a warning so much as it is a reassurance.

But I don't think it's necessary. The proof is in the next time you turn it on, of course.

And we always have to remember that people don't read messages. Each screen should try to communicate one and only one thing to the user--and even then it'll get ignored a lot of / most of the time. So if we were to try to put two messages up (wait 30 secs to unplug; all seeds will be wiped), it becomes probably exponentially more likely that neither message gets read.

Longer term, I've been wanting to think through a "super-noob" onboarding mode that has almost everything stripped out of the UI and is more of a guided step-by-step experience. Like when you turn it on, it should only offer two choices: Set up a wallet or Create a new seed.

In this kind of mode, I think a standalone explainer screen on powerdown could work. It's not meant to be displayed every time (hopefully the user graduates from super-noob mode quickly-ish), but is part of the difficult onboarding process.

EverydayBitcoiner, 2022-12-25:
Oh, I kind of like the idea of a tutorial/walkthrough mode for first time users. Though I'm not sure about stripping out the UI considering the idea is to get the user used to the interface is. I would rather a new mode include explainer screens and prompts, but maintain the same UI otherwise. That said, I'm thinking that currently this would be a low priority project given that it's probably easier and more effective at the moment to simply make walkthrough videos and explainers.

On the topic of a shutdown warning I'm convinced it's not needed. Going to close out this issue as not needed, but may open up a low priority feature issue for a tutorial mode.

closed 2022-12-25 by EverydayBitcoiner