SeedSigner/seedsigner #289: [GUI] More consistency in formatting buttons

EverydayBitcoiner, 2022-12-27:
@easyuxd, tagging you because I know you are the UX guy.

A couple of things that I want to change, but not sure if there is a reason they are the way they are.

One is the right carat in the two images below. The carat seems extra and inconsistent and I'm not entirely sure what it is conveying because every menu option leads me to a place where I am presented with more options (I get that the advanced one is indicating a submenu, but if it is going to have the carat I think it should be the very bottom button). Is there a reason that these need to be here or can they be taken out to be more consistent?

[image]

I'm also a little bit bothered by the fact that only one button on the Seed Options list has an icon. I think it would be more consistent to either have all buttons one way or the other rather than a mix of some with icons and some without. Below is an example of them all having icons on that list.

[image]

easyuxd, 2022-12-27:
Good eye, @EverydayBitcoiner! I support these changes, but would look to @kdmukai for confirmation as he implemented this.

According to the pattern we established, chevrons (carats) should only be present when they navigate to a nested submenu. If a list item navigates directly to a task/flow, a chevron should not be present.

The Advanced submenu is a perfect example of where you should see one. And I agree that grouping these items at the bottom would promote findability.

—

Agreed that we should go all-or-nothing with the icons, otherwise it’s less scannable. Personally, I think it helps reduce cognitive load to remove the icons here, but if the icons provide a useful visual aid for users I have no objection to adding them.

EverydayBitcoiner, 2023-01-02:
> Agreed that we should go all-or-nothing with the icons, otherwise it’s less scannable. Personally, I think it helps reduce cognitive load to remove the icons here, but if the icons provide a useful visual aid for users I have no objection to adding them.

I think you may be right that it may be better to have no icons on this screen.

jdlcdl, 2023-01-27:
Personally, I dont have a 'strong' opinion on "w/ icons" or "w/o icons", but DO like consistency.

Maybe @EverydayBitcoiner has two branches like:
* issue_289_seedoptions_icons
* issue_289_seedoptions_noicons
... and could issue pull-requests for both so that we could take a look and then decide to merge one or the other?

As for the right-chevron on "Backup Seed", I too am not convinced it needs to be there (unless it also needs to be there for Address Explorer, Export Xpub and others which require more choices.