SeedSigner/seedsigner #290: Shuffle SeedQR

jrc-dev, 2022-12-29:
Folks, I wrote this, not wanting to be arrogant, but wanting to improve this amazing project, please take this as constructive criticism.

I made some assumptions:
If the equipment is stolen, you are safe.
If the QRcode is exposed, everything is lost. (we are avoiding liability).
If the QRcode is exposed, but it has Bip39 passphrase, it is safe, it can add some security, but it also adds complexity.

Have you considered using a shuffle algorithm to secure the SeedQR code? The option is to use Python to generate a secure random key or a deterministic key by equipment, and use this key as a seed for a shuffle algorithm like the Fisher-Yates shuffle. This will shuffle our QR code in a way that makes it useless if it is exposed or photographed( because the shuffle seed), and can be physically stored on SD card equipment. This can be a great way to protect user QrCode!

If the equipment is stolen and this key is extracted, and will, there is no risk. this key can even be generated and stay on the SD card clear text, it is only used to protect the SeedQRCode been exposed. If the QRcode is exposed, no risk without the seed to revert the shuffle.

Recoverability:
But now we lose Recoverability, as the QRcode no longer matches the user's seed. Yea! And this can be solved if the shuffle seed is created deterministically using the seedphrases... so same seed phrases generate same key to shuffle, same key can be used to recover from a SeedQr Code.

It's important to understand that now we keep our backup keys (which may be 12 or 24 words long) safe and secure and hidded, and that the QR code we now generate and use on a daily basis (called the SeedQr) will be different from the backup seeds...

Here is some example in JS
https://www.npmjs.com/package/seedrandom
https://en.wikipedia.org/wiki/Fisher%E2%80%93Yates_shuffle

EverydayBitcoiner, 2022-12-29:
This seems to be along the same lines as issue #218 and seems to be something that could be useful and improve security. I have just started tinkering around with SeedSigner so have not deeply thought through all of the different trade offs with these different setups, but would be interested to know what others think.

SeedSigner, 2023-01-02:
It's something that we've discussed, but it isn't on the roadmap. Adding a Word 13/25 passphrase is the functional equivalent of encrypting your seed, and doesn't require adding any additional dependencies or other code that we're not already using. I appreciate the suggestion and the idea.

NoSandMan, 2023-01-14:
I do see potential OPSEC benefits to encrypting the Seed QR prior to export. At present if your QR is exposed it is immediately clear it is bitcoin seed. However, if the QR was encrypted this is not necessarily the case.

jdlcdl, 2023-01-14:
I'm torn on this one, between:
* let a power user be a power user; if they don't need it, they won't use it.
and
* don't allow a new user to harm themselves when existing standards (like passphrase) exist.

kdmukai, 2023-01-14:
NACK.

My default position is to avoid any novel approaches to Bitcoin security that aren't already a well-established best practice. I am not a cryptographer and am not qualified to weigh in on whether or not a new use case is sound. If the broader community hasn't vetted it, I'm not comfortable going first.

What you're describing sounds like it's applicable for BIP-39 mnemonics in general (i.e. not particular to SeedQR security). In that case I'd suggest pursuing it as a Bitcoin BIP first (yes, I know, much heavier lift). If the actual cryptographers that work on Core and the rest of the critical, eagle-eyed reviewers like the approach, then we'd be on a much stronger footing.

bitcoinprecept, 2023-05-01:
NACK. I've had this same instinct and desire on my own timeline and raised the point to the seedsigner team separately. All else being equal, I agree with the OP that this is a superuser feature I'd like to have, BUT... @kdmukai convinced me by arguing that this idea deserves consideration as a BIP instead of a Seedsigner enhancement. 

If we're going to be scrambling QR codes, regular users should be able to scramble their 12 words too. And if their 12 words are scrambled through a defined Bitcoin standard, then there's no need for Seedsigner to implement a custom scrambling function. We can at that point scan the scrambled word list into the existing seedsigner software with no changes.

SeedSigner, 2023-05-01:
Going to go ahead and close this one. Not in the roadmap at this time for reasons cited above, appreciate the input.

closed 2023-05-01 by SeedSigner