SeedSigner/seedsigner #294: [BUG]: Cannot finalize seed w/o passphrase if editing passphrase

jdlcdl, 2023-01-05:
**Is there an existing issue for this?**

    I have searched the existing issues

**Current Behavior**

While finalizing a pending seed, if the user adds a passphrase then decides to edit that passphrase by deleting all characters -- effectively removing the passphrase, they will remain stuck in the "Add Passphrase" entry screen with no option to bail-out other than to finalize the seed with the unwanted passphrase, then discard it in order to start over.

Additionally: If adding a passphrase that consists only of space characters, an exception is raised that looks like:

System Error
(!)
ValueError
seed_screen.py, 1037, in post_init
range() arg 3 must not be zero

**Expected Behavior**

When finalizing a pending seed, if the user starts to add a passphrase and then edits that passphrase effectively removing it, this should be acceptable and there should be some indication that the fingerprint has not been changed; it has no passphrase.

Additionally: If a passphrase consists only of space characters, this should result in something other than an exception.

**Suggested Fix**

I do not have any suggested fix for seedsigner codebase at this time, but an investigation is required and I intend to follow up later.

As well, an investigation into the bip39 specification, its reference implementation, and other implementations is needed before I can form an expectation about how passphrases that consist only of space characters should be handled. At the same time, because seedsigner currently trims leading and trailing spaces from the passphrase, verification that this is standard practice is warranted.

**Anything else?**

This issue was also discussed in https://github.com/SeedSigner/seedsigner/pull/291

jdlcdl, 2023-01-05:
On the topic of "space" characters within a bip39 passphrase, whether they are leading, trailing, or fill the passphrase entirely...

My finding, so far, is that there is no special treatment to "spaces", as opposed to other characters, and that the untrimmed passphrase is appended to the word "mnemonic" and passed thru to PBKDF2 as the salt argument when creating the bip39 seed.  The passphrases ("", " ", "  ") will all create unique bip39 seeds and master hd wallets for the same mnemonic phrase.  Redundantly, passphrases ("a", " a", and " a ") will all create different wallets.

I am not surprised that nobody has complained about this, but I do consider this a bug that would result in the inability for someone to use seedsigner to recover a bip39-to-bip32-hd-wallet which was created with a passphrase that had leading, trailing, or nothing-but spaces.... assuming that the original ux did NOT also trim the passphrase before handing it over to a bip39 library.

References that I have checked, so far, are:

* https://github.com/bitcoin/bips/blob/master/bip-0039.mediawiki 
The bip39 specification has no specific mention of special handling of "space" characters for the passphrase.

* https://github.com/trezor/python-mnemonic
The bip39 reference implementation does appear to contain code that makes special treatment for spaces in a passphrase, and I have tested that leading, trailing, and nothing-but-spaces does in fact result in unique bip39 seeds.

* https://github.com/diybitcoinhardware/embit/blob/master/src/embit/bip39.py
The bip39 implementation within embit also does not appear to treat spaces in a special manner and I have tested that the bip39 seeds embit.bip39 creates are the same as the bip39 reference implementation above.

...more to come

EverydayBitcoiner, 2023-01-06:
Since the " character could be part of the passphrase as well it may confuse people. It may be better to use something like an under bracket. Here's a convo I found about exactly how to best display a space.

https://ux.stackexchange.com/questions/91255/how-can-i-best-display-a-blank-space-character

jdlcdl, 2023-01-06:
If I were a coding wizard, I'd have already re-implemented the screen where the passphrase is reviewed, and each and every character of the passphrase would be displayed similar to the keys in the keyboard component.  Otherwise, no matter which character we use to represent a space, it could get confusing.

A passphrase of: 
"That's what she said", he said

Would currently print exactly as it looks above, while if the user added one trailing space, it would read like:
""That's what she said", he said "

Yuck!

The optimal display, in my mind, is to use some hint that does NOT include wrapping-with or replacing any characters that might also be valid as part of the passphrase (UTF-8 can support over 1 million).

There is good news, however.  Open source development is a slow march towards "better" and this pr could take a long long long time to be merged, giving me a chance to figure-out the wizardry that I already know would be much better than my quick fix.

Thank you for your thoughts and for the link @EverydayBitcoiner, and for your issues template that helped me write the first post of the issue.  I'm a fan!

easyuxd, 2023-01-06:
Very thoughtful, @jdlcdl. I tend to agree with @EverydayBitcoiner that adding quotes on the Review screen could be confusing.

Certainly an edge case, but if leading/trailing spaces were supported it could warrant exploration of other visual affordances that don't conflict with the charset. I'm thinking use of secondary color (grey), maybe adding an underline or input box around the passphrase?

EverydayBitcoiner, 2023-01-07:
A few different ideas to throw out there:
[image]

jdlcdl, 2023-01-07:
Morning thoughts:

While utf-8 may support over one million possible characters it's not reasonable to believe the seedsigner keyboard can support them all, neither can most other human input devices.  In any locale the subset of probable chars in a passphrase is considerably smaller.

@easyuxd and @EverydayBitcoiner pointed out that double-quotes to wrap a passphrase for this edge-case may be confusing.  I'll admit it was a very anglophone assumption to make; a francophone might have made the assumption to wrap with << and >>; all of these characters, for wrapping text, are much more likely to cause confusion because they are more likely available to existing keyboards where a previous passphrase has been set.

@EverydayBitcoiner pointed me to a discussion where this topic is discussed, and just above has included some examples of replacements that might be appropriate as a solution.

My current dilemma is the classic bang-for-the-buck consideration: "Is it worth it?".

* I know, deep in my heart, that I'd like for every character in a passphrase to be displayed with a different background color, or a faded border just as @EverydayBitcoiner's examples or like the rounded rectangle of an enabled key in our keyboard component, but I also realize that where this code needs changed is currently implemented via appending to a centered TextArea, where I believe that my current options to do so are limited;  Wrapping or replacing the passphrase is a one-liner while refactoring the entire screen for this edge-case that has existed for some time but never caused any complaints feels short-term "costly" to say the least, even though it would be long-term "rewarding" to my toolbox.

* I know that I DO want to become capable of wizardry in this code base, but this will not come overnight.  Perhaps other wizards would have suggestions or commits to apply to the related pr.

Aiming for "Done!" or at the least "better", and taking @EverydayBitcoiner's hint to replace space characters with "\u2423" (somewhat like underscore) or other easily available utf-8 characters which are NOT readily available in existing human interface devices and improbable in passphrases seems like a compromise worth pursuing; I will pursue it.   New to github and knowing that this is my first pr ever, I will need to take a look at what I can do as far as pushing to the same branch that I've already issued a pr for in issue#295, I may end up just leaving examples here in an image as I've done above, then asking for your feedback again.

jdlcdl, 2023-01-07:
**The compromise I have in mind (see above) might read something like:**

This string has spaces, but none leading, trailing, or consecutive.

␣This␣one␣has␣a␣leading␣space.

This␣one␣has␣a␣trailing␣space.␣

And␣this␣one␣has␣␣multiple␣consecutive␣spaces␣␣␣within.

This string has single spaces, and the next string has 4 spaces only.

␣␣␣␣

jdlcdl, 2023-01-07:
**Alternatively, the blank white vertical rectangle "\u25af" would read something like:**

This string has spaces, but none leading, trailing, or consecutive.

▯This▯one▯has▯a▯leading▯space.

This▯one▯has▯a▯trailing▯space.▯

And▯this▯one▯has▯▯multiple▯consecutive▯spaces▯▯▯within.

This string has single spaces, and the next string has 4 spaces only.

▯▯▯▯

jdlcdl, 2023-01-07:
Many unicode characters that I tried, "'\u23b5" and many geometric shapes are not visible w/ current fonts in seedsigner.  The "\u2423" under-score-ish character looks "just ok" when other visible characters are next to it, but it's not obvious that it's a space when the passphrase is only spaces.  The left-7/8s block is the clearest one for me.  It doesn't join together when next to another space as a full block would, and it also doesn't look as there is the block AND a space as is the case when I tried the left 3/4s-block.  Is it a space, or is it an un-supported unicode character? that's my only concern for confusion.  As an anglophone, I almost prefer wrapped with double-quotes.  ho hum :(

PR #295 (2023-01-17): Issue 294 passphrase
fixes #294
While finalizing a seed:

* If user flows into adding a passphrase and then decides to edit -- only to delete it, the next step is back to Finalize w/o a passphrase, instead of being stuck having to finalize a non-empty passphrase only to discard the seed and start all over.

* If user has leading or trailing spaces in their passphrase, ever since v0.4.3 this would have resulted in deriving the wrong hd wallet because "passphrase.strip()".  With this change, the next step is to review the passphrase, which will be double-quoted so that they can see the space cha [... cut]
closed 2023-01-17 by newtonick