SeedSigner/seedsigner #296: [Chore] Update `embit`; refactor BIP-85 flow to use `embit`

kdmukai, 2023-01-08:
BIP-85 support added to `embit`: https://github.com/diybitcoinhardware/embit/pull/35

Included in the latest `embit` release.

We should update to the latest `embit` and remove our BIP-85 implementation in favor of calling the implementation in `embit`.

jdlcdl, 2023-01-09:
I have not been watching that repo long enough to form an opinion on historical "stability" of embit api's as soon as a new feature is added... but I notice that it's in pypi since last week. 

Normally, I'd say "There is no prize to be won for being the very first to add stability to someone else's api." and would rather let others go first, because the cost of not doing so is that your own api is also not stable... in our case, it just means making local changes at worst case, I see that embit's bip85 mnemonic app passes the tests that it needs to pass, that every other lib passes.

kdmukai, 2023-01-09:
Stepan is pretty quick about updating Specter-DIY and the Specter team generally takes up `embit` updates into Specter Desktop quickly.

But regardless, part of the chore here is to review his changes and assess possible impact, as you've done/begun to do. @jdlcdl can you provide a link to the specific `embit` line number that caught your eye?

As for "visibiilty", that should be PR'ed out!

jdlcdl, 2023-01-10:
> can you provide a link to the specific embit line number that caught your eye?

It's a harmless/petty style-ism that caught my eye, where I dont notice similar whitespace use in other lists/list-comprehensions within that file (and haven't noticed it elsewhere in that code base) -- as well as a space/comma all at line 18 in src/embit/bip85.py.

https://github.com/diybitcoinhardware/embit/pull/35/files

Had I noticed it before merge, I might have left a comment on that line... or I might have just parsed it in my head and thought: "harmless".

As for the "visibiilty" typo on 3 lines in our codebase, I was waiting for the chance to review any pr that touched settings_definitions and the settings view, so I could add a comment like: "boyscout rule: check it in cleaner than it was checked out while peers' eyes will be looking at it... typo on word 'visibiilty'" instead of fixing something that isn't broken.  I don't want to condition fellow devs to NOT peer review by asking them to peer review something that isnt really a fix.

kdmukai, 2023-01-10:
> As for the "visibiilty" typo on 3 lines in our codebase...

As a former English teacher, I consider any PR that fixes a typo as an immediate ACK.

My first PR in a Bitcoin Core BIP was to amend an errant apostrophe: https://github.com/bitcoin/bips/commit/99ca1c887686f3881db6e1206d9442f775545400

jdlcdl, 2023-01-10:
I've left my thoughts/questions about embit/bip85 for Stepan, but not sure if I've left them in the right places.

https://github.com/cryptoadvance/specter-diy/issues/72
and
https://github.com/diybitcoinhardware/embit/commit/41f0a38f83f9df77bee7e0ddece4b32e3d935879

closed 2023-01-25 by newtonick