SeedSigner/seedsigner #309: [Feature] Add an option to generate a new seed via coin flips

kdmukai, 2023-01-22:
Coin flip entry can use the keyboard UI w/custom keys to just "H" and "T".

Probably makes sense to break the task into chunks of 11 flips (each word = 11 bits) and flip for one word per screen.

For human stupidity reasons, I would not display the resulting word while flipping (reduce the odds of people fudging their flips so they can hand-pick their favorite word).

Flow:
* Tools -> (coin icon?) New Seed
* 12- or 24-word seed?
* Word #1 (keyboard UI w/just very large "H" and "T" keys; display output as "H" and "T" but convert to 1/0 under the hood). Include note that heads = 1, tails = 0). After 11th flip, enable a "Next" button or something.
* Word #2... Word #11...
* Final Word: forward straight to the existing Final Word screen that supports coin flips for the checksum calc. Alter the explainer screen so that coin flips are the ONLY option (instead of listing options, just replace with a "Next" button).
* Flow from that Final Word screen then takes over. Checksum word is calculated, mnemonic is displayed, user is given the option to do the quiz to test that they wrote their mnemonic down correctly.

Options/Bonus: Maybe enrich the final mnemonic word display to include the raw bits for each word?

mister-1-wizard, 2023-02-04:
Thanks for this, Keith. Let me know if I there is anything I can do to help (I'm new to Github and fiat job is crazy busy at the moment, but will be glad to learn/help if no time pressure). 

Here are some additional thoughts I had (please feel free to ignore what is impractical; goal in mentioning now is to try to leverage effort -- get most bang-for-buck -- while still in brainstorming stage):

(1) For faster and for better entropy with coin toss method, user may throw 11 coins at once (mentioning here in case affects UI/overall flow)

(2) I wonder if simultaneous with this it might be possible to work XOR into flow; for example, user loads two seeds (e.g., one generated from image and one generated by dice or coin toss) into first two seed slots, button press causes XOR of seeds in slots to be written to slot 3

(3) Where user wants to do (2) recursively, user moves XOR output at slot 3 to slot 1, generates new seed at slot 2, and presses XOR button again to write XOR of slots 1 and 2 to slot 3

kdmukai, 2023-03-05:
@mister-1-wizard for your (2), if we have coin flips as an entropy source to begin with, I don't see much value in an XOR option built on top.

If you want the "good" entropy of coin flips, use coin flips to generate a new seed. Full stop.

So to me, (2) is a completely separate issue (implement SeedXOR), which I remain skeptical/hesitant on.