SeedSigner/seedsigner #319: Navigate keyboard jumps to available letters

jwander14, 2023-02-09:
When entering a seed and inputting a word, it would be nice to skip over the greyed out letters and jump directly to the closest potential letters based on the direction of the joystick. 

Example: enter b and everything is greyed out except a, e, i, l, o, etc. If you click down, would be nice to jump directly to i instead of h. Click right, jumps to e instead of c...

Thanks for considering. PV 🤙

kdmukai, 2023-02-09:
@jwander14 There may be / probably are ways around the complications, but the basic challenge is described here:

https://twitter.com/KeithMukai/status/1412878350406889476

jwander14, 2023-02-09:
Not sure I follow...if you have 
XOX
OXO
XOX
...and you are in top left x, I would think a joystick right moves to the far right x in top row and/or a joystick down from either x in the top row goes to the middle x (assuming the o's are greyed out). Another joystick down from middle x would default to bottom left x...then quick right joystick to move to bottom right x. An up from top row goes to bottom left x. (Sorry I'm misunderstanding why this couldn't work...not a programmer but do know a change can have so many downstream consequences!)

Does offsetting every other line work?[image]

kdmukai, 2023-02-09:
That's inconsistent.

>...and you are in top left x, I would think a joystick right moves to the far right x in top row 

So that rule is: for horizontal movements, STAY in the current row, skip "O"s until the next "X" in same row...

>and/or a joystick down from either x in the top row goes to the middle x (assuming the o's are greyed out).

But the rule is different then for vertical movements: CHANGE columns as needed to land on the nearest "X".

And whatever rules you set, there's always an awkward, crappy result: by the above rules, navigate from top left X to bottom left X (ignoring wrapround navigation):
```
X O O O
O O O X
O O O O
O O O O
X X X X 
```

Total sequence ends up having to be: D, D, L, L, L. You can get there but you end up having to go through an unexpected, not totally intuitive path.

In most real-world cases I think some set of keyboard jump rules do end up in fewer directional inputs. But it's these weirdo cases that drive me nuts.

That all being said, I do remember now the UI improvement that would make these jumps a bit more intuitive. It's hard to explain, but imagine two semi-transparent rulers--one vertical and one horizontal. They're both overlaid on the keyboard to indicate your current horizontal and vertical position. You'd still have weirdo paths for any possible ruleset, but the jumps would be easier to visually track.

SeedSigner, 2023-02-10:
If feasible, we could also implement both options and let users choose in settings which they prefer. Not sure if "option creep" is a thing though.