SeedSigner/seedsigner #337: Design Files for open_pill_mini_w_coverplate

fti7, 2023-02-25:
Hi,
very nice Enclosures! 😀
Any chance that you will commit/opensource the Design files for the open_pill_mini_w_coverplate as well?

Thx!

SeedSigner, 2023-02-27:
The printable files are already available here. Good luck with the print, I’d recommend doing the controls in TPU if you’re able to.

https://github.com/SeedSigner/seedsigner/tree/dev/enclosures/open_pill_mini_w_coverplate <https://github.com/SeedSigner/seedsigner/tree/dev/enclosures/open_pill_mini_w_coverplate>

@SeedSigner

fti7, 2023-02-28:
> The printable files are already available here. Good luck with the print, I’d recommend doing the controls in TPU if you’re able to. [...]

Already found that Files and printed it. It looks very nice, even with PLA Controls 😄 
I wanted to ask if you could publish the Design Files (.f3d Autodesk Fusion) like on the first Open Pill Enclosure as well.
Would like to play around with it and make some modifications (e.g. A flat Nintendo Controller Style "Cross Key")

Thx

SeedSigner, 2023-03-01:
I’m open to this. In the interim, here is the fusion file. Ignore the lid that is still aligned with the main chassis — the one off to the side is the correct current one.