SeedSigner/seedsigner #338: Verification documentation/steps break on v0.6.0

dadofsambonzuki, 2023-02-26:
The [Verifying the Software](https://github.com/SeedSigner/seedsigner#verifying-the-software) instructions no longer work on v0.6.0.

jdlcdl, 2023-02-26:
Thank you for submitting this issue:

A solution to this is currently proposed as pr #285 

Until then, assuming you trust the sha256 file (because you've verified the sha256.sig signature), you can:

`shasum -a 256 -c seedsigner.0.6.0.sha256`

And look for an ": OK" appended to the pi02w.img line in the output of above.

newtonick, 2023-02-28:
@dadofsambonzuki the README instructions have been updated. Can you give these steps a try and provide feedback? https://github.com/SeedSigner/seedsigner#verifying-that-the-downloaded-files-are-authentic-optional-but-highly-recommended

dadofsambonzuki, 2023-02-28:
Verified. Works like a charm and clearer than before.

newtonick, 2023-02-28:
Thanks for the feedback!

closed 2023-02-28 by newtonick