SeedSigner/seedsigner #343: [Tests] Improve `embit_utils` test coverage

kdmukai, 2023-03-05:
Now that #342 has been merged, this should be a pretty straightforward PR.

Current coverage report:
```
Name                                Stmts   Miss Branch BrPart  Cover
---------------------------------------------------------------------
seedsigner/helpers/embit_utils.py      59     48     40      0    11%
---------------------------------------------------------------------
```

I recommend writing tests into a new tests/test_embit_utils.py file.

jdlcdl, 2023-03-05:
I'd played with this https://gist.github.com/jdlcdl/79392ec1417863c49498815edc7e01bd a while back, so I'll look to continuing towards more coverage.

jdlcdl, 2023-03-06:
> I'd played with this https://gist.github.com/jdlcdl/79392ec1417863c49498815edc7e01bd a while back, so I'll look to continuing towards more coverage.

simply adding that file to a new branch and running coverage again brings the report to:
```
Name                                    Stmts   Miss Branch BrPart  Cover
-------------------------------------------------------------------------
src/seedsigner/helpers/embit_utils.py      59     27     40      3    52%
-------------------------------------------------------------------------
```
...but I have no intention of leaving it like this.  This was originally my attempt at exhaustive coverage of a single function.  Seeing all that has evolved in your latest testing sprint, I'll convert my style from test functions to test classes and methods and do my best to adopt similar style to remain consistent with the rest of the tests.

PR #388 (2023-07-20): Improve embit utils test coverage 343
resolves #343

At this point in time, this pr does NOT add any new application code, it is aimed at maximum test coverage of embit_utils, bringing coverage from ~11% to ~95%.

This pr does however hint towards 2 "good-practice" changes in embit_utils; see coverage's html report for details:
* appending an `else` at the bottom of get_single_sig_address() and raising a ValueError if script_type unknown.
* appending an `else` at the bottom of get_multisig_address() and raising a ValueError if policy is unknown.
closed 2023-07-20 by newtonick