SeedSigner/seedsigner #344: Feature Request: reminder to pull SD after boot

jevidon, 2023-03-06:
With the new OS allowing for SeedSigner to be run without leaving the microSD card in the device, it would be a nice idea to flash a reminder on screen once the user can safely remove the SD card and proceed with their use.

Marc-Gee, 2023-03-12:
Agree. I'm going to see what I can do on this. Perhaps reminding the user just before they create or scan a seedQR.

**Potential issues:** 

1. The Restart command (currently) needs the SD inserted. I will investigate why.
2. The settings menu will need the SD inserted for saving persistent settings.

SeedSigner, 2023-03-15:
I see two options for when we could potentially insert this kind of notification

1 - right after the splash screen, once the main menu menu has loaded
2 - the first (or any) time that the user goes to create or load a key into the signer

Like we do with some of these more potentially intrusive features, ideally we could have an option to disable this option if users preferred not to see it. Open to more thoughts on the above.

dtonon, 2023-07-19:
I was investigating about this possible weakness and I'm happy to know that it is possible to remove the microSD!

I support the @SeedSigner proposal, and talking about preference I would also add one that forces the user to remove the microSD after the boot to go on with the use.

jdlcdl, 2023-07-21:
Since we can remove the microsd, it's almost insane NOT to remove the microsd.  But I'm sure there are many enclosures already in the wild which don't allow access to the microsd card without disassembly.  Still, I like the idea of many reminders... so that we can instill good habits... and inspire our users to take a dremel to their no-micorsd-access enclosures.

dtonon, 2023-07-21:
> it's almost insane NOT to remove the microsd

Agree. In fact I would set it as the default mode with the option to bypass the block.

jdlcdl, 2023-07-21:
Ever since 0.6.0 I've wondered how this works.

Seedsigner has gets "confused" whenever you pull the microsd before the main app gets started/restarted.  It's because the main app doesn't know about the current status, it only learns of status when the microsd is inserted or removed.

Since a solution would involve both seedsigner-os and seedsigner repos, it will certainly be appreciated to have input from @newtonick and @DesobedienteTecnologico who may already have more natural ideas on how to go about this.

My gist, a work in progress since this morning:
https://gist.github.com/jdlcdl/94d4baccb053af769e64b83a8d8cbe2a

jdlcdl, 2023-07-22:
The link just above ^^^ Is my draft pull-request to solve this issue.

tl;dr Just one warning at seedsigner startup.  Must click "Continue" to make it go away.

jdlcdl, 2023-07-23:
The initial pr commit would show a screen like:
[image]

Some feedback I've received from the telegram group include:
* custom icon and a different color
* instead of at boot, only when loading seeds.

On the first point, w/o changing color, I've chosen the MICROSD icon and reworded the warning:
[image]
...in order to have your feedback on how this warning might be "received" by users.

DesobedienteTecnologico, 2023-07-23:
> Ever since 0.6.0 I've wondered how this works.
> 
> Seedsigner gets "confused" whenever you pull the microsd before the main app gets started/restarted. It's because the main app doesn't know about the current status, it only learns of status when the microsd is inserted or removed -- and only once it's listening (as seedsigner app starts).
> 
> Since a solution could involve both seedsigner-os and seedsigner repos, it will certainly be appreciated to have input from Nick and DT (who I'd previously tagged before realizing a solution might NOT include seedsigner-os), they may already have more natural ideas on how to go about this.
> 
> My gist, a work in progress since this morning: https://gist.github.com/jdlcdl/94d4baccb053af769e64b83a8d8cbe2a

Until this PR https://github.com/SeedSigner/seedsigner-os/commit/a644a3e38722984f740e99ccafbff4c936ced0e3 / https://github.com/SeedSigner/seedsigner-os/issues/20 we had the option to umount the MicroSD manually and don't have also the "Dirty Bit" https://github.com/SeedSigner/seedsigner-os/issues/37#issuecomment-1419781001 . Things were addressed in another way.

In short. If the splash screen is going to be created, will be great that once you press the Continue button it umount the MicroSD, so the "Dirty bit" thing can be addressed and keep having reproducible builds even directly from MicroSD as @jdlcdl commented https://github.com/SeedSigner/seedsigner-os/issues/37

dtonon, 2023-07-24:
> ...in order to have your feedback on how this warning might be "received" by users.

I like the microSD icon, it is visually nice and immediately clarifies the subject of the message.
About the interface I would suggest, following the two cases:

---

**A) Optional removal**

_MicroSD is still inserted!_
< icon >
_The microSD will not be used anymore, we suggest to remove it before loading the seed for maximum security_
[ I removed it ]
[ Keep it mounted ]

---

**B) Mandatory removal**

_Remove the MicroSD_
< icon >
_For maximum security you must remove the microSD to continue_
[ I removed it ]

---

Case A should address the issue of persistent settings; I don't use it so I don't know how the internals work.

Marc-Gee, 2023-07-24:
> I like the microSD icon, it is visually nice and immediately clarifies the subject of the message. About the interface I would suggest, following the two cases:
>
> **A) Optional removal**
>
> _MicroSD is [...]

Hi, 
I also like the micro SD icon. 👍

@dtonon - you raise an excellent point: optional or mandatory?
I like mandatory, as it leaves the user in a more secure position. Secure by design. There appears to be no downside as the code deals with saving persistent settings if they later want to use that. 

So: case A gets my vote. However, the language should allow for potential use of (later dev of PSBT's on a Micro SD), so I suggest 
" For maximum airgap, you must remove the microSD._ [ I removed it ]"

Power users could leave it inserted to see what happens. Which they will of course try , but only with their fake seeds - bc even power users will want to be reminded to remove the SD when dealing with their real seeds. 

Thanks for your clarifying contribution. 
M

Marc-Gee, 2023-07-24:
> > Ever since 0.6.0 I've wondered how this works.
> > Seedsigner gets "confused" whenever you pull the microsd before the main app gets started/restarted. It's because the main app doesn't know about the current status, it only learns of status when the microsd is inserted or removed -- and only once it's listening (as seedsigner app starts).
> > Since a solution could involve both seedsigner-os and seedsigner repos, it will certainly be appreciated to have input from Nick and DT (who I'd previously tagged before realizing a solution might NOT include seedsigner-os), they may already have more natural ideas on how to go about this.
> > My gist, a work in progress since this morning: https://gist.github.com/jdlcdl/94d4baccb053af769e64b83a8d8cbe2a
> 
> Until this PR [SeedSigner/seedsigner-os@a644a3e](https://github.com/SeedSigner/seedsigner-os/commit/a644a3e38722984f740e99ccafbff4c936ced0e3) / [SeedSigner/seedsigner-os#20](https://github.com/SeedSigner/seedsigner-os/issues/20) we had the option to umount the MicroSD manually and don't have also the "Dirty Bit" [SeedSigner/seedsigner-os#37 (comment)](https://github.com/SeedSigner/seedsigner-os/issues/37#issuecomment-1419781001) . Things were addressed in another way.
> 
> In short. If the splash screen is going to be created, will be great that once you press the Continue button it umount the MicroSD, so the "Dirty bit" thing can be addressed and keep having reproducible builds even directly from MicroSD as @jdlcdl commented [SeedSigner/seedsigner-os#37](https://github.com/SeedSigner/seedsigner-os/issues/37)

It makes good sense to unmount it once the user responds, and that will release the dirty bit, this putting the hash back to it's original value. Win-win! 

M

dtonon, 2023-07-24:
@Marc-Gee 

> I like mandatory

Me too, in fact as I said some days ago I would make it the default option with a "security first" approach. I don't see any downside.

> So: case A gets my vote

You mean case B, right? :)

>  ... the language should allow for potential use of ...
> "For maximum airgap, you must remove the microSD"

Absolutely agree 👍

jdlcdl, 2023-07-24:
On the topic of mandatory: 
I'm aware of some enclosures that do not allow access to the MicroSD when assembled.  Maybe a release where it's optional, and if they haven't taken a dremel/drill to their enclosure before the following release, then they'll have been warned?

On the topic of unmounting the MicroSD to clean the dirty-bit:
I actually like the habit of pulling the microsd as soon as its LED goes dark (3-5s after power-on).  When this is the case, the seedsigner app has not even started yet so no chance to unmount politely.  We have a thread watching for asynchronous pulling of the microsd by the user w/o doing anything in the menus, they just yank it... so in any case where the user doesn't want to do anything special except yank the microsd, we'll never have a chance to clean the dirty bit.  Also, about powering off, we say "Just Unplug It!", so that the user knows it's safe to pull power at any time (maybe because they're being surprised/attacked/interrupted) and I think it would be an aweful loss of this feature, teaching them to hesitate/rethink when they should just yank.
My preference would be for seedsigner-os to build the image with the dirty-bit already set, and then if they never altered persistent settings, release-hash and their microsd hash (at least for the bytes written to the microsd) wouldn't change.

Thank you for your suggestions, I'll be leaving screen shots of what they would look like.

jdlcdl, 2023-07-24:
I've tried to make alternative warning screenshots for the suggestions I received since yesterday.
In some cases, I wasn't able to implement the exact textual suggestions (w/o reducing fonts) so I took it upon myself to reword phrases... hopefully without destroying the "spirit" of the message.  But you'll have to let me know. 

---

Keith shared this text suggestion (and I made a guess for the title and colored warning-text).

[image]

---
@dtonon suggested both an Optional and Mandatory warning behavior... without any warning-text, which plays quite nicely with the SDCARD icon that doesn't leave enough vertical space underneath.

If Optional:  It was suggested to have 2 buttons "I removed it" and "Keep it mounted", but I chose simply to leave the "Continue" button as the only choice, because it's easy, and because we could use the removal of the microsd as an event equal to pressing "I removed it".

[image]

And if Mandatory: It was suggested to have 1 button "I removed it", but because we can know if it was removed and continue on that event, I altered the button text to reiterate what they need to do.

[image]

---

@Marc-Gee suggested slightly different terminology.  I have altered the button text to "Continue" because we can know if the microsd was removed and continue on that event (whether optional or mandatory) or let them press "Continue" if optional and they want to leave it in. 

[image]

---

Thank you for all of your suggestions.  Keep 'em coming!

SeedSigner, 2023-07-24:
Agree with this last part, about manually setting the dirty bit during build. Though hashing the memory card after booting it is not described or recommended anywhere in our documentation, so the specific workflow of hashing the memory card seems less relevant to general users. With the tiny size of our deployment image, for users it makes sense to just hash the image on their computer (if they have concerns) and re-deploy that image to the memory card. Is there any importance to not having the dirty bit set beyond the hashing use case?

SeedSigner, 2023-07-24:
My 2 sats for this:

Title:
MicroSD Inserted

Elaboration:
Removing the MicroSD card
before creating or loading seeds
is recommended.

I'm a stickler about line breaks but not sure if the above would need to be further broken down.

dtonon, 2023-07-24:
@jdlcdl 

>  It was suggested to have 2 buttons "I removed it" and "Keep it mounted", but I chose simply to leave the "Continue" button as the only choice,

Effectively two buttons are not necessary at all.
Maybe the text in this option could be:

**MicroSD inserted**
_It is not needed anymore, you can remove it before proceeding for maximum security._

It works well both at boot and before the load of the seed.

jdlcdl, 2023-07-24:
@SeedSigner made these suggestions, but "seeds" on the second line wrapped to another... so despite being a stickler for line breaks, I adjusted it below.  It would actually read fine... just... on an undamaged waveshare (I'm missing 9 pixel columns at last check... it's degrading gracefully sofar).

[image]

jdlcdl, 2023-07-24:
> @jdlcdl
>
>
> Effectively two buttons are not necessary at all. Maybe the text in this option could be:
>
> **MicroSD inserted** _It is not needed anymore, you can remove it before proceeding for maximum [...]

I tried as:
```
It is not needed anymore,
you can remove it before proceeding
for maximum security.
```
...but "proceeding" wrapped to another line.  4 lines doesn't read so bad, but the wrapped line break was not ideal.  If you suggest more with exact line breaks, I'll try.   It seems that 25chars is roughly max per line, but never know until we try it.
I changed your suggestion to:

[image]

dtonon, 2023-07-24:
@jdlcdl 

>  It seems that 25chars is roughly max per line

Thanks for the info, I was just counting them to make more appropriate suggestions that take line breaks into account :)
I'm not a native English speaker and I don't know how your version with -- sounds, I would write:

_It is not needed anymore.
Remove it, then proceed,
for maximum security._

Marc-Gee, 2023-07-24:
Thanks for your reply. Solid analysis. 
Optional and then mandatory, now sounds correct path to me. I did not consider the various cases, Especially the early ones. 

That's a very pretty icon you drew! And which looks a lot like the state of Oregon.... 🕵️‍♂️ , So I will side with the old, but clunky SD card icon until [our global tech overlords update their iconography  sets and], we show the world what a modern Micro SD card looks like.

Marc. 
 

Wow, thinking outside of the boxes box today! 
"My preference would be for seedsigner-os to build the image with the dirty-bit already set"
🤛😁

Marc-Gee, 2023-07-24:
> On the topic of mandatory: I'm aware of some enclosures that do not allow access to the MicroSD when assembled. Maybe a release where it's optional, and if they haven't taken a dremel/drill to their enclosure before the following release, then they'll have been warned?
> 
> On the Icon: This was the best I could find. We're limited to icons already in the system. This was from a FontAwesome icon set. We also have a SeedSigner custom icon set. If I knew how to make icons, I'd add one to our SeedSigner custom icon set which was a microsd, instead of a standard sd card... and the contacts would not be visible, and the height of the icon would not go all the way to the bottom of it's space where it conflicts with the warning text, and it would be oriented horizontally with the irregular edge on top so that it is in the same orientation as it would be when they're pulling/inserting it. Just putting it out there that I think we can do better with the icon... I just don't know how to do that, yet. Not quite this, but as an idea: [image]
> 
> On the topic of unmounting the MicroSD to clean the dirty-bit: I actually like the habit of pulling the microsd as soon as its LED goes dark (3-5s after power-on). When this is the case, the seedsigner app has not even started yet so no chance to unmount politely. We have a thread watching for asynchronous pulling of the microsd by the user w/o doing anything in the menus, they just yank it... so in any case where the user doesn't want to do anything special except yank the microsd, we'll never have a chance to clean the dirty bit. Also, about powering off, we say "Just Unplug It!", so that the user knows it's safe to pull power at any time (maybe because they're being surprised/attacked/interrupted) and I think it would be an aweful loss of this feature, teaching them to hesitate/rethink when they should just yank. My preference would be for seedsigner-os to build the image with the dirty-bit already set, and then if they never altered persistent settings, release-hash and their microsd hash (at least for the bytes written to the microsd) wouldn't change.
> 
> Thank you for your suggestions, I'll be leaving screen shots of what they would look like.

Thanks for your Solid analysis @jdlcdl ! 
Optional and then mandatory, now sounds correct path to me. I did not consider the various cases, Especially the early ones. 

That's a very pretty icon you drew! And which looks a lot like the state of Oregon.... 🕵️‍♂️ , So I will side with the old, but clunky SD card icon until [our global tech overlords update their iconography  sets and], we show the world what a modern Micro SD card looks like.

Marc. 
 

Wow, thinking outside of the boxes box today! 
"My preference would be for seedsigner-os to build the image with the dirty-bit already set"
🤛😁

jdlcdl, 2023-07-27:
Nick had originally mentioned a different color.

I've played with adding a MICROSD icon to the seedsigner-custom-icons set which is a microsd instead of a standard SD card, instead of vertical with contacts facing, it's oriented similar to what a microsd would look like when removed from a pi0... and it does not interfere with text on the following line.

This is just a sample of a different color.  The pulsing border is red, the icon is red, the warning text beneath is red, everything else remains the same (white title, white text, yellow button).

We can keep playing w/ ideas
[image]
, eventually it will be time to decide (hopefully before the next release).

jdlcdl, 2023-07-27:
On the suggestions that this screen should ONLY appear when loading seeds, I will now ramble:

For what purpose does a user power-on their seedsigner, if they are not about to load seeds?

I'm biased that we should teach our users best-practices as "habits".  If this screen will only appear when they've failed to remove the microsd card between 5s after power-on and the time it takes to boot and splash... and they've taken good habits, then they'll never see it.  If they choose to ignore it, we've warned them once per session, but we won't harp on them many times if they back-out half way through loading/creating seeds (and it would be pointless to continue harping on them if they ignore it and continue loading more seeds; it's already too late).  If they cannot take action, because their enclosure does not enable this, same situation; it's a reminder, only once per session, to modify/replace their enclosure if they really care about best practices.  Further, this ability to remove all storage before loading secrets is something that sets us apart from other hww projects, so it may also be seen as "Once per session marketing" only to those who need it because they haven't yet realized what sets SeedSigner apart from the rest of the crowd.

If the first thing they want to do is to change a persistent setting... they'll be reminded when in persistent settings that they need to re-insert the microsd card... or they'll have left it in because "they know better" and will ignore this one-time warning.  With good habits, they won't see it, they'll change setting, verify that it's persisted... and this is where they'll ideally re-insert the microsd, enable persistent settings, and remove it immediately thereafter.

Otherwise, we'd have to protect:
* Scan: because they might load a SeedQR and it would be too late to warn them after seedsigner sees it.
* Tools: for all actions except Address Explorer via loading a multisig wallet descriptor.
* Seeds: for all actions.
(but it would not be needed for any of the Settings actions).

The implementation to check almost everywhere, that would lead to less code would be to "decorate" View classes which lead to loading/creating seeds.  But this would still be many classes to decorate and to make sure they're all working properly.  We're very careful NOT to write anything except settings to the microsd card, so we'd be adding lots of code just to warn the user about protecting themselves against something we don't even do... yet.  Beyond that, decorated classes of this form, other than the @dataclass is not even a pattern that we currently exploit... so we'd be taking a new convention with lots of code changes for something that I've argued above is already not ideal.  I'm not even convinced that using decorators is a code-convention pattern that we'd benefit from yet (they're neat, but it's easy to take them way too far, where it gets out-of-hand for readability/debugging).

I'll argue the above point at the meeting tomorrow; so get your ducks lined up where I've missed something.  I'm always open to being convinced to change my mind.

jdlcdl, 2023-07-28:
Some push back on my thoughts to leave this as a StartUp warning were heard at today's meeting, and resulted in another commit to pr #410.

Changes are:

User will see this if microsd inserted and leaving MainMenu for Seeds, Scan or Tools.
A flag is set that they've been warned, and they won't see this warning again... unless they re-insert their microsd which will reset the flag to warn them next time.
If they've disregarded it in the past and loaded seeds, no matter what they do, since it's too late, they will not be warned again this session.

jdlcdl, 2023-08-01:
And one more, with a new icon proposed by a design expert.

fyi: Title Text, Sub-Icon Text, and Body Text are still undecided.  
Please let your voices be heard as we approach release!!!

[image]

jdlcdl, 2023-08-01:
@SeedSigner also had suggestions.  Knowing that he's a stickler for new-lines, I played with 2 variations.

[image]
[image]
[image]
[image]

dtonon, 2023-08-01:
@jdlcdl 

> but would "Pro Tip" be a better title?

I don't know if a title of this kind is really useful, because it is not so informative. "opsec" could be an obscure term for someone, "tip"/"advise" remind me to something useful but optional, while as [you pointed out](https://github.com/SeedSigner/seedsigner/issues/344#issuecomment-1645557706) there is no need to keep the microsd inserted at all. I would simply stick with the fact in the title (the microsd is inserted), and clear instruction as main text (remove it before continuing). Or for consistency the title could be "Caution", as in other alerts.

> with a new icon proposed by our design expert.

I like it! But I would rotate 90° (more space, better readability).
Talking about the color, I would pick the same red used in the backup seed warning, and apply it also to the button/flashing border. A single color should be less confusing and does the job well.

jdlcdl, 2023-08-01:
@dtonon 

Thank you for your interaction and suggestions over the past 2 weeks on this issue.

I'll be sure to verify that your input has been taken into account by other devs and especially the maintainers.

I think there are multiple preferences desired by different users.  Some cannot remove their microsd card because their enclosure does not allow access, others will never see it because they have good habits, and new users might be alarmed as a "first impression" which is that last thing we want.  Balancing these interests and finding a fit for all users is most important, and it's turning out to be less than obvious.

I expect to make whatever changes the maintainers ask for, and I'm also open to an all-together different notification implementation which has been suggested recently.

Thanks again!

easyuxd, 2023-08-01:
@jdlcdl 

Late to the party, but I have some feedback on the content:

We have 3 colors for messaging: green (success), yellow (warning), red (error). Red is scary and this isn't technically an error, so this one should be yellow to remain consistent with the other dialogs.

If the message is for noobs, maybe further simplifying the title would help?

Security Tip
< Icon >
MicroSD is inserted!
For maximum security remove the MicroSD card before continuing.
[ Continue ]

Since there's a MicroSD icon, the colored "MicroSD is inserted!" line may even be redundant and might just amount to additional reading for users.

Aside: There's so much great thinking in this thread that would be great to capture as guidelines for future messaging. I previously documented [content strategy best practices](https://github.com/easyuxd/seedsigner-design-system#content-strategy) (mainly related to navigation) that might prove useful.

jdlcdl, 2023-08-02:
@easyuxd 

Thank you for your input.   I should have sent an invite long ago; I'm thankful that Nick did.

I've tried without the colored "MicroSD is Inserted!" sub-icon text, which leaves more vertical space despite which icon is used.  Here are two samples, the first with the existing FontAwesome SDCARD as well as your new MICROSD sent recently.

[image]

[image]

dtonon, 2023-08-02:
@jdlcdl 

> Thank you for your interaction and suggestions over the past 2 weeks on this issue.
> I think there are multiple preferences desired by different users. 

Thank _you_ for taking my point of view in consideration :)
Yes, of course, it is essential to take into account the most frequent preferences/use cases to optimize the overall usability.

@easyuxd 

> Since there's a MicroSD icon, the colored "MicroSD is inserted!" line may even be redundant and might just amount to additional reading for users.

I agree. The following "removal" prompt explains the state of the microsd by itself.

> would be great to capture as guidelines for future messaging.

If you need some help I would be happy to contribute to the UI/UX.

> [content strategy best practices](https://github.com/easyuxd/seedsigner-design-system#content-strategy)

Great starting point!

dtonon, 2023-08-02:
I was trying to recreate a template UI (central) to test some ideas:

[image]

and then, alerted by @easyuxd nick, I went down the rabbit hole and found his https://github.com/easyuxd/seedsigner-design-system repo, that's absolutely great! Is it the official design guide? (Should be!)

We could use his template to quickly create a matrix of cases/personas and the relative ideal messages, then make a synthesis of the needs and pick up the best one.

jdlcdl, 2023-08-02:
@dtonon

I don't know the entire history well, because I arrived in late September of last year.  I believe that the UI rework was done for version 0.5.0 so it would have been the months leading up to that release, early 2022.   I know that easyuxd is often tagged for user-interface and design expertise, but I didn't feel comfortable enough that I could request his time whenever re-using existing components for a similar task... but I was wrong, I should have asked for his advice up front.

I do believe that the link easyuxd shared is the official design guide... I'd been pointed there in the past and forgot all about it.  For icon/font styles, you might check out the 2 font sets [here](https://github.com/SeedSigner/seedsigner/tree/dev/src/seedsigner/resources/fonts), the OTF's are the only ones I believe which are in use, but I could be mistaken.

I like your samples.  The only thing I can say about the middle one is that it might lead me to put the microsd card into the pi0 upside down (but it would be perfect for my pi2... which is).

easyuxd, 2023-08-02:
@jdlcdl @dtonon 
Big fan of the great work you guys are doing here. I have a [v0.5.0 Figma file](https://www.figma.com/file/YCvGdWbyhxPrYqxk8gZWuA/SeedSigner?type=design&node-id=10602%3A10718) (I'm in the middle of cleaning it up) that you can use if you want to test content or icons. Just duplicate the file so you can edit it. Tag me anytime -- I'm always happy to help, I just don't always see the GH issues.

kdmukai, 2023-08-08:
Alternate implementation approach underway in #424.

[image]

jdlcdl, 2023-08-08:
My damaged waveshare is getting worse, and my phone camera always stunk,

but there's a new screenshot generator in town, with some new icons thanks to @easyuxd.
pr #410, likely to be replaced by #424, would look like this:

[image]

PR #424 (2023-08-15): Reminder to remove SD card via Toast notification
Alternative to #410 but builds upon behind-the-scenes updates from that branch. Would close #344 upon merge.

Display a toast notification popup on the Main Menu if the SD card is still inserted.

As has been thoroughly, arduously(!) discussed, my preferred presentation for this notification is to simply let the user know that they *can* remove the SD card, but omit any mention of security or warnings.

The precursor PR, #410 has evolved into an optional Must Remove hard stop which can work in conjunction with this notification:
* Default: no h [... cut]
closed 2023-08-15 by newtonick