SeedSigner/seedsigner #349: Feature Request: No screensaver when showing flickering QR Code

petzsch, 2023-03-15:
Hi,

I've just been playing around with a very large transaction (167 inputs) and noticed that there are some problems, most related to my webcam scanning this (changing focus and stuff that I need to take care of)

However the seedsigner showing a screensaver every so many minutes has become a real problem.

Could you disable the screensaver when showing a QR code that is flickering?

Would be much appreciated. :-)

SeedSigner, 2023-03-15:
Wow -- 167 inputs is quite a few but could me more common than people think with regular DCA transactions and UTXO consolidation. 

This is of course a reasonable request -- I've also had the screensaver come up when I was transcribing a SeedQR which was less than ideal. What might make the most sense is to extend the time before the screen saver comes on when a user is engaged in either of the above activities. If a user happens to leave a device unattended for many 20-30 mins while displaying PSBT QRs, or if they were to get interrupted during SeedQR transcription, we may still want the screen saver to come in eventually at some point to prevent burn-in.

I'm curious with your webcam if you have a static resting place for the signer, or are you holding it up to the webcam the entire time? If you're holding it up and routinely do these kinds of large transactions, you might consider a setup where you can secure the SeedSigner in a fixed position. My favorite setup for these larger transactions is to use my phone as a tethered webcam (now native on MacOS, and apps like EpocCam also facilitate this) and position both the phone and signer in static positions during transfer. If you hadn't seen it, there's also a setting for high QR density in the signer's advanced settings that will reduce the amount of frames that need to be passed back through your webcam.

Thank you for raising this, it's something we should definitely address.

petzsch, 2023-03-16:
> if you have a static resting place for the signer, or are you holding it up to the webcam the entire time?

I had put the seedsigner and the camera on my desk looking towards each other. Still the camera (Brio 4k Stream) kept on loosing focus, everytime I had reset the screensaver. In that case I had to point the camera at something far away and then back at the device and it would keep focus. As said, that is something I need to look into, maybe the logitech software which I hadn't installed thus far, allows manual control of such "features".

Just FYI, I also tried using the seedsigner with BTCPay Server and ran into some [issues](https://github.com/btcpayserver/btcpayserver/issues/4777). I hope that the team gets them sorted out. :-)

Love the little devices... really awesome. ❤️

SeedSigner, 2023-03-16:
Thanks so much for opening that bug report with BTCpay. I've attempted to sign with SeedSigner in their UI and it's been on my to-do list to to open an issue. With the browser interface, our signer seems like a perfect fit for when their users want to withdraw funds.

kdmukai, 2024-09-20:
Bump! This is still an issue in v0.8.0.

@newtonick can you tag this for v0.9.0?

cc: @bitcoinprecept

kdmukai, 2024-09-20:
First cut at some updated rules to debate. Screensaver activates after...:
1. Animated QR: 10 minutes? Never?
2. Static QR: 2 minutes (no change)
3. SeedQR transcription UI: 2 minutes (no change)
4. Camera live preview: Never (already implemented, I think)

1.) I could go either way. Maybe leaning towards "Never", but with an asterisk explained more below.

2.) Seems uncontroversially fine to me.

@SeedSigner noted that he ran into 3.) when stamping into steel. I think the screensaver kicking in is a mild annoyance in that case but not really disruptive.

4.) Must remain "Never"; any time the camera is on, user must be aware of it.

---

*(As a separate TO-DO, I think it's worth considering a UX timeout option (ON by default) to protect the user in the worst-case scenario where they happen to walk away in the middle of 3.) and it's still running (actually, similar risk if it's abandoned at ANY point when a seed is loaded). If the max timeout is reached, the UX would completely wipe / reset / power down)