SeedSigner/seedsigner #354: Feature Request: Load BIP-85 child seeds at creation

FractalEncrypt, 2023-03-26:
When creating a BIP-85 child key, after verifying the seed words of an indexed parent key, there is no option to export the seed to QR, or to load the key into memory. Currently the user must manually key in each letter of each word of a child seed, making the process much more cumbersome. 

Feature Requests:
1. Load child keys into memory like the parent seed.
2. Display child key fingerprint at time of child seed creation

jdlcdl, 2023-05-16:
Without sharing my thoughts on whether this is a good idea or a bad idea,  I'm playing with this foot-gun over [here](https://github.com/SeedSigner/seedsigner/compare/dev...jdlcdl:seedsigner:load_bip85_child_seed).

It works like so:
* If you've used BIP-85 child to view the child's mnemonic, then you'll be offered the existing choices "Verify" and "Skip" as well as a new choice "Finalize child".
    * If you choose "Verify" and you succeed in verification of the child mnemonic, then "OK" will bring you to the finalize step where you can "Done" or "Add BIP-39 passphrase" to load the new seed.
    * If you choose "Skip", then you'll be brought back to the Seed Options menu for the original parent seed... just as is the current behavior.
    * If you choose "Finalize child", you're brought directly to the finalize step where you can "Done" or "Add BIP-39 passphrase" to load the new seed -- even though you never verified a good backup... because you don't need it because that's what BIP-85 is for... but you best remember which is the parent, if it's 12 or 24 words, and the exact index!!!
    
* This branch is a minimalist edit to existing views which are widely used whenever creating a new seed or verifying seed words via backup.  However, the above changes SHOULD NOT be offered in these other non-BIP-85 use cases... so we'd need to be careful testing other flows which use these views... before considering to merge this foot-gun.

more later, maybe.

gpatkinson, 2023-07-16:
@jdlcdl is the load_bip85_child branch functional over on your fork?

jdlcdl, 2023-07-16:
> @jdlcdl is the load_bip85_child branch functional over on your fork?

I believe that it is functional -- for testing, though I doubt anyone has looked at it.  I wouldn't ever recommend that anyone run a non-official release for "real".  It appears to be 51 commits behind this repo's "dev" branch;  I'll get it updated soon.

gpatkinson, 2023-07-16:
@jdlcdl cool. I think this feature will be a great addition to the code base.

bitcoinprecept, 2023-08-08:
I didn't see this issue before creating my own ([Issue 426](https://github.com/SeedSigner/seedsigner/issues/426)). I went ahead and closed mine to defer to this one.

This is a functionality I would personally like to see. With it, BIP-85 becomes attractive to me.

jdlcdl, 2024-03-05:
In the telegram group, a user has expressed a use case like "a super seed to derive many online hot-wallets", within the past hour.

FractalEncrypt, 2024-03-11:
> Without sharing my thoughts on whether this is a good idea or a bad idea, I'm playing with this foot-gun over [...]
 words via backup.  However, the above changes SHOULD NOT be offered in these other non-BIP-85 use cases... so we'd need to be careful testing other flows which use these views... before considering to merge this foot-gun.
> 
>     * I can imagine foot-gun proponents making a strong argument that "viewing the child seed words" is not even necessary, and that "Finalize child" should be offered as soon as inputs: "12-or-24 words" and "index" are known.
> 
> 
> more later, maybe.

I have built jdlcdl's branch and have it loaded on a seedsigner device right now.

It works great and loads the BIP-85 keys for use in the device like the parent keys. BOSS!

comments:
When I create the child key by selecting the numerical index, it brings me right to a screen with 4 options (but it didn't show the seed at all first)
Verify
Review
Skip
Finalize Key

if I click "Verify" - well since I haven't seen the words yet - there is no way I can actually complete this step (and there is no back/exit option)
I entered a wrong word, and it did give me the option to review the words now.

If I click "review" - it works great, and I can then verify and finalize the key

If I click "Skip" it does not load the key, it just brings me back to the page for the parent seed. if I go back to look at my in-memory keys, the new key does not display. So from your description, this sounds like it is working as expected.

"Finalize Key" also works exactly as described.

Bravo this is super cool AF.

jdlcdl, 2024-03-11:
Thank you for reviewing my branch.

You bring up an important point that is most likely due to my laziness it not storing the fact that "Review" had been previously selected and simply offering enough (and in this case too many) to get the job done -- the dirty way.

Please feel free to leave a pointer in the telegram group, from time to time, to see if there is interest for other users and devs.  Also, please understand that in the end, the current maintainers (SeedSigner and Nick) as well as others need to make the best (often most prudent) decision about new features -- for the long term best interests of all users and the project itself, even if some users really-really want a particular foot-gun.

Thanks again FractalEncrypt!  I'm grateful for your review.

FractalEncrypt, 2024-03-11:
My pleasure, thanks for doing the hard work! I'm embarrassed it took me this long to get around to actually testing out the very thing I asked for. 

My plan to attract interest is to do a Twitter thread showing people what this is and why I think it is cool. Then we'll see if anyone agrees or if this is just me being weird, which is always a distinct possibility.

jdlcdl, 2024-03-13:
@FractalEncrypt While you're working on a twitter thread, I invite you to see one my recent posts in the seedsigner telegram group, where I try to describe the hazards of a bip85 puzzle.  That tg post is https://t.me/c/1469623205/51772

FreeOnlineUser, 2024-03-25:
I have built my first OS from the seedsigner manual installation instructions, and implemented this bip-85 branch on a pi zero w. Thankyou for the work that has been done. I'm very impressed with the functionality, it was definitely the missing piece in my security model. I'd say it's perfectly usable now. Management of the active child keys could become the next challenge, I imagine an index next to each child fingerprint could assist c0 c1 etc but not at all necessary.
fwiw I did the build in qemu emulator on win11, and wrote the image to sd with balena.

jdlcdl, 2024-04-05:
In the telegram group this morning, a past-contributor has asked about this feature.

aheinzer, 2024-08-27:
In my opinion, this is a very useful feature and I have been waiting for it for some time.

Unfortunately, this issue has not made it into the current release v0.8.0.

What is the best way to support this issue so that it can be implemented? My knowledge of Python is still very limited.

jdlcdl, 2024-09-13:
A request for similar functionality was requested in the main telegram group yesterday afternoon.

Meisterzunge, 2024-09-13:
Hey devs, I would love to see this feature on main branch!
In my opinion this will definitely ease the handling of BIP-85.

szeco, 2025-04-25:
Catching up with the latest discussions on this topic in the telegram group.. feels like this feature has been intentionally shelved for now. Personally I would find it useful, but also understand the concerns and risks a bit more after reading the chat. 

Still, the current implementation of BIP85 support feels a bit half baked. Why allow the user to generate the seed but not load it? If the core devs feel like it's a bad idea, why not remove it from the build completely? 

The idea where you have the option in settings to have BIP85 disabled by default, but also enable creation and loading the child seeds independently seems the best compromise. 

Hope this gets a bit of attention in the next release.

21M4TW, 2026-09-07:
Yes it is really unfortunate that BIP85 seeds still cannot be loaded. Is their any objection to support this?

FreeOnlineUser, 2026-09-10:
I think it’s included in the main branch now, just enable it in settings, and turn on persistence to keep it enabled.