SeedSigner/seedsigner #355: Tools -> Address Explorer -> Scan wallet descriptor: "Not Implemented Yet"

oren-z0, 2023-03-27:
Happens for a descriptor of a single-seed wallet.

Is this entire feature not implemented yet?
Is it just my specific wallet? My derivation path?
Is it the QR code format?

jdlcdl, 2023-03-28:
If you are scanning a single-sig native-segwit or single-sig nested-segwit wallet-descriptor qrcode which you got from:
specter desktop, Wallets, Settings, Export, Go to export details,
then you are doing it right and it IS the case that this is not yet implemented.

It does however work for multisig wallets of those script types, which then allows for exploring addresses w/o your private seed.

What you might use as an alternative, until this has been implemented, is to scan an address qrcode so that seedsigner can verify that it's yours (but this is assuming that your private seed will be unhidden while doing so). 

Also, you might use the address-explorer on seedsigner and compare those to what specter-desktop gives you (same assumption as before PLUS quite heavy on the eyes).

Lastly... do you trust that specter-desktop has NOT been compromised since you initially setup the wallet?  I'm guessing that we should not and that's why you're checking.  The only way to do this is to temporarily/securely unhide/expose your private seed to seedsigner and let it be the one to verify each address... or to have verified and printed out addresses at wallet setup time.

I hope this info was helpful.  Maybe other devs will have better alternatives.

oren-z0, 2023-03-29:
> do you trust that specter-desktop has NOT been compromised since you initially setup the wallet? I'm guessing that we should not and that's why you're checking.

Yes, this is exactly what I wanted to check, to calm down my paranoia... I could install a new watch-only wallet, enter the same descriptor/xpub and see that the addresses are the same - but doing this with SeedSigner would be simpler. I think this feature should be prioritized over multisig wallets.

Thanks!

oren-z0, 2023-11-26:
@jdlcdl Any update on this?

jdlcdl, 2023-12-24:
@oren-z0 I do not have an update on this at this time.

But I still agree that to be able to scan a single-sig wallet xpub/descriptor and access address viewer would be a nice/desirable feature.   I will remember this issue in case there is any progress to that end and will include a pointer so that you'll see it here.