SeedSigner/seedsigner #356: JGMTx: when signin txs elaborate the full hex ready to be broadcasted

pony-montana, 2023-04-01:
Whats would be the seedsigner signin behaviour:
1) Scan for an unsigned tx.
2) Review and sign the tx.
3) Shows the qr of the signed Tx hex ready to be broadcasted.

Having this functionality will make a lot more flexible the use of the seedsigner. It could be activated by a check in the menu option or creating a new possibility under "scan psbt".

Assumptions:
A lot of people (like me) just want to use a signin device to not expose a private key on an internet connected device. A simple one-key-wallet is what a lot of people would use; multisig is something absolutely interesting and useful, but only few people want to manage that.
Creating a simple flow for the " Just Give Me the damned Tx" or JGMTx will permit to be compatible with  an absurd amount of software that could act as "coordinator" on every platform. Isn't that standardization?

My personal problem with seedsigner:
I have an old librebooted thinkpad and use sparrow as a view-only wallet. But my old thinkpad doesnt have a camera for scanning qr. 
Clearly solutions could be a lot: 
- Buy a camera for the thinkpad.
- Use the smartphone as external camera connecting it with some hacky and buggy solution.

But imagine if I could do JGMTx in this way:
1) Create the unsigned Tx on sparrow on my thinkpad.
2) Scan the qr with the seedsigner.
3) Review and sign the Tx.
4) Scan from seedsigner the full signed Tx hex on my android phone and broadcast it. 

More simple and KISS. More flexible for a lot of people living in country where could be difficult have access to a pc with camera or a high-powerful smartphone to run bloated softwares like bluewallet.

SeedSigner, 2023-04-04:
I like the concept in that it is consistent with SeedSigner as a kind of Swiss Army Knife for bitcoin. What is missing is a simple QR encoding & decoding desktop/mobile app that could translate pasted hex into a parse-able animated QR set, and then intake an animated QR set and dump it to copy/paste-able text. Given that we're a volunteer project this may require someone that is passionate about the concept to pick up the ball and run with it.

pony-montana, 2023-04-04:
I hope I will be able to contribute more, I'm trying to learn :). 
From seedsigner-os what is missing is the ability to elaborate and export the full tx signed (and not the sign). For example, I think the HW foundation passport is able to shows the signed tx as the last qr, like in the video: https://www.youtube.com/watch?v=whtsRLmvkf4 (I dont have it, so i'm not 100% sure of what I'm saying).
Simple bitcoin wallet doesnt works with seedsigner (tested and reported here https://github.com/nbd-wtf/obw/issues/47), but I think it would only require to show the signed tx for seedsigner to become compatible (and speedup compatibility with a lot of other software wallets).
I really love seedsigner and only want it to become even more flexible to interact with :).

pythcoiner, 2023-05-30:
> I like the concept in that it is consistent with SeedSigner as a kind of Swiss Army Knife for bitcoin. What is missing is a simple QR encoding & decoding desktop/mobile app that could translate [...]

I made such a tool for testing purposes with my seedsigner, (both simple and animated QR), I can polish it and make it more distributable (it is actually just a python script using PyQt) if there are some needs.

SeedSigner, 2023-05-30:
Thank you for this message. What I like about this approach is that this kind of desktop tool could make SeedSigner immediately compatible with other additional wallets. From a security perspective, I think it makes sense that our project does not publish the tool -- too much of a security concern for us to make the signer software AND software that is installed on your desktop/laptop computer. Is your work available currently in a repo?

pythcoiner, 2023-05-30:
> Is your work available currently in a repo?

I'll make one, just made the tool Saturday for play with miniscript stuff with my seedsigner...
I will ping you there so you can have a look

pythcoiner, 2023-05-30:
you can check [here](https://github.com/pythcoiner/SeedQReader)
quite basic for now, a lot to be done

SeedSigner, 2023-05-31:
Took a peek and it looks like a great start! Do you think you'll be able to implement the UR standard for animated QR codes?

pythcoiner, 2023-05-31:
> Took a peek and it looks like a great start! Do you think you'll be able to implement the UR standard for animated QR codes?

yes, of course, just list what you think must be done, I'll try to handle that