SeedSigner/seedsigner #361: ZeroCam

OliverOffing, 2023-04-29:
Do we know if the [ZeroCam](https://thepihut.com/products/zerocam-camera-for-raspberry-pi-zero) is compatible with the seedsigner? They're both "5MP 1080p" so I'm optimistic. Wondering what the difference is between that and the OV5647 version, and whether someone's build an enclosure for this model.

OliverOffing, 2023-04-29:
[This article in German lists the ZeroCam](https://hey-bitcoin.de/anleitung/seedsigner-hardware/) so I'm guessing it works.

OliverOffing, 2023-04-29:
[@SeedSigner himself (herself?) has provided a prototype design for a case for the ZeroCam](https://twitter.com/SeedSigner/status/1621670808128045062). I can try and print this once I finish building the hardware.

bitcoinprecept, 2023-05-01:
From a software perspective, yes. This is confirmed by Seedsigner within the community Telegram group as recently as this month (see screenshot) as well as by various contributors.

[image]

From a hardware perspective, it depends on which enclosure you're going for. Not all 3D printable enclosure designs fit with each camera model:
https://github.com/SeedSigner/seedsigner/tree/dev/enclosures

newtonick, 2023-05-23:
Thank you @bitcoinprecept for answering the question!

closed 2023-05-23 by newtonick