SeedSigner/seedsigner #363: [Enhancement] Create UI element to explain screen brightness adjustment technique during QR presentation

bitcoinprecept, 2023-05-12:
Seedsigner has a hidden feature to adjust screen brightness from any QR screen: joystick up or down to adjust brightness on any QR screen.

It would be helpful to add a brief UI popup or some other indication within the UI to inform users of this function.

pythcoiner, 2023-05-30:
> It would be helpful to add a brief UI popup or some other indication within the UI to inform users of this function.

maybe a screen displaying "Use joystick Up/Down for adjusting brightness" every x seconds, or every x frames for dynamic QR?

SeedSigner, 2023-05-30:
Open to someone taking ownership of this feature; a brief overlay for a few seconds as soon as a QR code is displayed could be a viable solution. The message can't be persistent, to prevent it from interfering with the QR code transfer, but brightness is a common issue that people run into and we aren't communicating that the brighter/dimmer option is available anywhere in the UI/UX.

PR #389 (2023-07-20): Add QR brightness tips screen
resolve #363

[363#issuecomment-1568754458](https://github.com/SeedSigner/seedsigner/issues/363#issuecomment-1568754458)
closed 2023-07-20 by newtonick