SeedSigner/seedsigner #367: Using SeedSigner with 3.5 inch LCD touchscreen on Raspberry Pi 3B

leafyus, 2023-05-21:
Hey there,

I am trying to run on a RP3B with a MHS 3.5 inch RPi Touchpanel. So far I am only seeing the shell output on the screen, not the actual GUI of SeedSigner. 

Is this touchpanel even supported? Could I give inputs with touch gestures?
If not, is there any way I could alter the original code (I am not a pro coder) to make this work?
[image]

SeedSigner, 2023-05-22:
While I would like to get to the point where we can support a touchscreen of this size/profile, currently our software is hard-coded for that Waveshare 240x240 display. The RPi 3 is 100% fine, but you need the waveshare hat. While it is technically possible to adapt our UI to support the screen you have, it will be a complex process to support an additional display profile that we will have to give a fair amount of strategic thought to. I appreciate your interest in our project.

jdlcdl, 2023-05-22:
I love that there is interest toward a screen that myself and my generation can actually read.  Whenever we get around to this, it will be a game-changer for accessibility.  Thanks for sharing your interest, you're not the first and won't be the last to want similar.

leafyus, 2023-05-22:
Thanks for the replies! I might dig a little deeper into the code if I have the time, but the waveshare hat will suffice until then :)

stn021, 2023-06-11:
Same here. I have a waveshare 3.5inch RPi LCD (A) with 480 × 320 hardware resolution.
This one : https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(A)
It is a 40pin hat.

The 240x240 hats are out of stock everywhere.

I would like to suggest a "quick solution". Support other display-sizes, perhaps simply as a line in config.txt, but only use 240x240 of the display. So, for now, don't try to scale up the display, just wrap it to the existing screen.

kdmukai, 2023-06-11:
@leafyus: Those sorts of screens are meant to essentially redirect/capture what would have been the hdmi output, which is produced at the operating system level. You can think of SeedSigner as an app that directly controls a separate display (NOT the OS hdmi out) that the OS otherwise doesn't even know about.

It is possible to wire up different screens in a way that SeedSigner can control them, but it is not trivial and requires both some decent coding chops AND a good amount of physical wiring knowledge (gpio pins and all the associated complexities of which pins can do what).

---

@stn021: It's much more difficult than that. The display that's integrated into the Waveshare hat uses a specific hardware display driver: ST7789. Our code is fairly easily compatible with any display using that driver. But larger displays (beyond 320x240) step up to something like an ILI8496 driver. Code to interface with the different hardware drivers is surely out there somewhere, but would have to go through a bunch of R&D to get integrated into SeedSigner.

kdmukai, 2023-06-11:
@stn021 see: https://twitter.com/KeithMukai/status/1419474349405388804

Unfortunately this effort was abandoned when I hit a brick wall trying to deal with the XPT2046 touchscreen interface. I could either get the touchscreen working or the display, due to... SPI(?) timings. I think I have a better idea for how to do it now, but it just hasn't been made a high priority. And my long-languishing work to port to esp32 would make all of that much easier to take on once it's done.

leafyus, 2023-06-11:
@stn021 actually I found that 240x240 on Amazon for a fair price. I’m located in Germany tho..

SeedSigner, 2023-06-12:
Link for Ali Express Waveshare hat for anyone that is having trouble sourcing:

https://a.aliexpress.com/_mKLi8ji

stn021, 2023-06-12:
> @stn021 actually I found that 240x240 on Amazon for a fair price. I’m located in Germany tho..

Yes, I know how to use google and amazon search. None of the well known german suppliers have this type of display. In that case it is a good idea not to experiment with some unknown supplier, as I found out in a few similar cases. If no well established supplier has it, then it can safely be considered out of stock.

leafyus, 2023-06-12:
@stn021 alright mate, didn’t wanna step on your toes

314159-r, 2023-11-06:
@kdmukai I kind of assume that any display with a ST7789 driver and 240x240 resolution would more or less work as a drop-in replacement (disregarding mechanics  and wiring).

But how about a 240x320 ST7789 display? Like the [Waveshare 2inch 240x320](https://www.waveshare.com/2inch-lcd-module.htm)
Does accepting some dead area, and only make use of 240x240 pixels make things easier?

Looking for larger displays with ST7789... The only 240x240 display I found is a Waveshare 1.55" display, but the difference from the current 1.3" isn't worth it considering it lacking joy-stick and buttons.
Their 240x320 2.0" display above is on the other hand is noticeable larger. If only using a square  surface, the seize would correspond to 1.7". 
Some fast trawling on Aliexpress gave some larger possible alternatives like these [2.4"](https://www.aliexpress.com/item/1005006133348282.html) and [3.2"](https://www.aliexpress.com/item/1005005443349049.html) displays. But all larger displays have 240x320 resolution (except for some 1.54" clones). 

So, if we want a larger display, everything comes down to if a 240x320 display is possible to adapt with reasonable effort..?

SeedSigner, 2023-11-06:
Appreciate these questions. I'll just throw in that the biggest motivation to get to a larger display is to improve the efficiency of the QR exchange process, so any screen where one of the dimensions is "240" won't accomplish this. Because of the need for accurate touch when using a virtual keyboard, we'll also want the touch tech to be capacitive (not resistive). From waveshare's offering, the one below seemed to make the most sense to me (all of their SPI touch screens appear to be resistive):

https://www.waveshare.com/2.8inch-dpi-lcd.htm

314159-r, 2023-11-06:
I certainly wouldn't mind higher resolution as well, but as mentioned above, that isn't an easy task. And I didn't expect that to happen any time soon. 
So I wanted to see if something can be done with what we have...

In my opinion, there is also a big motivation to have a larger screen, that simply is larger by size. So that some of us can read the content without putting the device under a microscope...  ;)

rafaelpac, 2024-12-30:
Guys, would it be possible to emulate the Waveshare HAT screen?
Something like this: https://www.bluetin.io/displays/simulate-oled-lcd-display-pc-pil-opencv/
If we manage to do that, we could run the 3.5 screen of the original post (with some xfce window manager).
It could be nice for development purposes too.

rafaelpac, 2025-01-09:
Hi! I got this to work!
It is a Raspi 3B+ (old [RaspiBlitz](https://github.com/raspiblitz/raspiblitz), not powerful enough anymore for today's nodes).
It is still very buggy.
I am running Bookworm (Debian 12, #430).
I flashed the regular latest release of Raspi OS with desktop.
Played around with all the dependencies...
picamera is totally deprecated in Debian 12, they have this new [picamera2](https://github.com/raspberrypi/picamera2), with all sort of different methods, I will need to refactor all the camera stuff yet.
The buttons are not behaving very well, maybe some refactor of the debouncing needed.
The rendering is done with OpenCV (cv2.imshow(cv.FULSCREEN)), so resolution doesn't matter.
Check video: https://photos.app.goo.gl/CtjpCG2VUaLdB5i2A

Master plan: put it on a [GamePi43](https://www.waveshare.com/product/raspberry-pi/portable-gaming/gamepi43-acce-us.htm) undercover as a SNES emulator!

kdmukai, 2025-01-09:
@rafaelpac very cool!!

After we get v0.8.5 released, I'll be finishing off my support for 320x240 screens: https://github.com/SeedSigner/seedsigner/compare/0.8.5-rc1...kdmukai:seedsigner:ili9341_320x240