SeedSigner/seedsigner #375: Suport cipherseed to bitcoin private seed

yywill, 2023-05-27:
Lightening using a diffrent seed algoritm other thant BIP39 
see: https://github.com/lightningnetwork/lnd/issues/2735

electrum and lnd use different seed formats. electrum uses BIP 39 while lnd uses a cipherseed. see https://github.com/lightningnetwork/lnd/tree/master/aezeed for more info on lnd's seed format

want to: caculate cipherseed of LND, and get bitcoion control back!

kdmukai, 2023-05-30:
Interesting.

1. Even with control of the seed, what would happen next? Manually execute channel closes, etc? Is there enough tooling out there to do whatever next steps you want to do? And does that tooling support QR-based signers like SeedSigner?
2. Support for aezeed is something I'd look to PR into `embit` to be a more generally useful solution (vs coding support just for SeedSigner).

emplexity, 2024-04-21:
Same expectation here. Would it be possible to get a BIP39 version of the seed created by LND?