SeedSigner/seedsigner #376: Scanning PSBT from Bluewallet stuck at 99%

Syturio, 2023-05-27:
My SeedSigner can't scan the QR Code from Bluewallet PSBT directly because it has too high density.
If i screenshot the QR and transfer it to my computer and fullscreen it, SeedSigner is able to scan up to "99% compete", but it never reaches 100% no matter what angle or how zoomed it's. It's frustating because i can't sign anything.

bitcoinprecept, 2023-05-28:
This definitely could be a bug. I might suggest engaging the community telegram support channel to get real-time troubleshooting engagement on this issue you're facing. From there we can determine how best to update this issue.

https://t.me/joinchat/GHNuc_nhNQjLPWsS

newtonick, 2023-05-28:
> My SeedSigner can't scan the QR Code from Bluewallet PSBT directly because it has too high density, my phone has a good monitor and high resolution, it seems that SeedSigner camera that can't pick up high density QR Codes.
> 
> I tried to screenshot the QR code and open it on my 27'' inch monitor at full screen, and SeedSigner is able to scan it but only up to "99% complete", it never reaches 100% no matter what angle or how zoomed it's, so i can't complete the transaction... I'm completely stuck. What am i suppost to do?

Does it work on medium density but not high?

newtonick, 2023-05-28:
> My SeedSigner can't scan the QR Code from Bluewallet PSBT directly because it has too high density, my phone has a good monitor and high resolution, it seems that SeedSigner camera that can't pick [...]

Can you try scanning BW into Sparrow? (Lee wet me know if you need instructions) 

Or if you’re comfortable with it, you can share with me privately the PSBT QRs to see if I can debug.

bitcoinprecept, 2023-05-30:
@newtonick solved this in the Telegram community. Issue can be closed. Details below:

OP: ... When i tried to scan from my PC, i only imported 1 QR, that might explain why it was stuck at 99%. I will try scanning all later. But the problem still remains, my SeedSigner cant scan those high-density QRs directly from my phone screen Bluewallet. I have a Samsung A7 2018, with 2k something pixel resolution OLED sharp screen, yet my SeedSigner cant scan those high-density QRs from it.

Nick: Sometimes the pi cameras are not in focus when you buy them. There are bunch of articles online about tuning them a bit. Here is the first one I found: https://zpjiang.me/2020/05/28/picamera-adjust-focus/

OP: Worked! I just had the rotate the camera with my fingers, and now the scan has way more focus and sharpness, im now able to scan high density QR codes

closed 2023-06-03 by newtonick