SeedSigner/seedsigner #390: Formalize strategy around handling of in-built Raspberry Pi OTP (One-Time Programmable Memory Block)

bitcoinprecept, 2023-06-14:
Nostr user [Nosthor](https://iris.to/npub14stash4guqdz0ftnqj5f8thuxv2m220mcxukf67vduc6lah5gjssykwwjt) identified a potential threat vector for seedsigner worthy of discussion and potential mitigation. They [write](https://iris.to/note1s9xlqmumlmy7cegqq6q8myqkst9nqjpmlquc9mr7wfg8uhgfpy4skdul5e): 

There are some tools that allow writing a private key to the inbuilt One-Time Programmable memory of the Raspberry Pi. Is there a way to break or disable the OTP memory permanently in order to rule out the (even remote) possibility of a private key being written there when using a SeedSigner?
[@SeedSigner](https://iris.to/npub17tyke9lkgxd98ruyeul6wt3pj3s9uxzgp9hxu5tsenjmweue6sqq4y3mgl)​ ​ [@KeithMukai](https://iris.to/npub1tv8gmfhalwnxxquxjzeh6gtdsdz6vg7vx0s3rt7s7uuw6aujh32qn77wn2)​ [@newtonick](https://iris.to/npub1rfd0hxdzcze6pzj29thuz34vur57wm9quje7w3edxjgusq6m47csnl7wrt)​ [@DesobedienteTecnológico](https://iris.to/npub1dtmp3wrkyqafghjgwyk88mxvulfncc9lg6ppv4laet5cun66jtwqqpgte6)​​ 

More info below...

OTP memory:
[github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/raspberry-pi/otp-bits.adoc](https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/raspberry-pi/otp-bits.adoc) 

rpi-otp-private-key tool:
[github.com/raspberrypi/usbboot/tree/master](https://github.com/raspberrypi/usbboot/tree/master) 

rpi-derive-key tool:
[lib.rs/crates/rpi-derive-key](https://lib.rs/crates/rpi-derive-key)

Some preliminary discussion from Seedsigner developers @DesobedienteTecnologico @kdmukai @jdlcdl:
"First thought: wouldn't hurt to write noise into any blocks big enough to store a private key."
"Second thought: If we wrote the SeedSigner public signing key into a register, would that help with any further "it's a legit software release" assurances? (I don't think so since the SD card can always have code that pretends to verify against the burned-in key)"

"Filling it with with zeroes would be my preference. Just in case."

[image]

"re: raspi video core otp.  Looks like maybe on registers 36-43 are writeable, and if you want to make them "used" then 0xffffffff would seem better.

I was able to write to mine, but it's turning 0s into 1s on a per-bit basis.  afterwards, looks like you can overwrite it (as long as you're only turning 0 bits to 1bits), but if you get to all ffs, they'd be used-up."

jahangir13, 2023-06-14:
Info from Raspberry for OTP:
https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#industrial-use-of-the-raspberry-pi
--> can be locked (also see the whitepaper pdf).

As far as I understand that it's about a seedsigner developer mis-using this? There are maybe many placed where 'unwanted' code could be injected during the build and compile process.

jdlcdl, 2023-06-14:
On reading https://pip.raspberrypi.com/categories/685-whitepapers-app-notes/documents/RP-003611-WP/Using-the-One-time-programmable-memory-on-Raspberry-Pi-single-board-computers.pdf:

Is it possible that this was not public info until first published in Jan 2022?  Even while my pi2 from 2014 seems to have always had these 58 registers?

I've tried the following on my RPi2:

Taking a look at what these registers "appear" to be storing:
`vcgencmd otp_dump >~/visible_otp_registers.txt`
They're indexed 8-66.  Most are unset, but visible exceptions are:
```
8-15:00000000

16:00280000
17:1020000a # Bootmode register
18:1020000a # Copy of bootmode register
19:ffffffff
20:ffffffff
21:ffffffff
22:ffffffff
23:ffffffff
24:ffffffff
25:ffffffff
26:ffffffff
27:00005050
28:a8f3cef0 # Serial number
29:570c310f # One’s complement of serial number
30:00a01041 # Revision code; also bits to disable overvoltage, OTP programming, OTP reading

33-35:00000000

36:43:00000000 # customer OTP registers were all zero, but flipped them all below.

44-66:00000000
```

Disabling the Customer OTP registers 36-43 via:
`/opt/vc/bin/vcmailbox 0x00038021 40 40 0 8 0xffffffff 0xffffffff 0xffffffff 0xffffffff 0xffffffff 0xffffffff 0xffffffff`
...and I ignored the errors, because it seemed to alter `vcgencmd otp_dump` as expected and could read these values.

Tried to read the internal private key (supposedly stored in registers 56-63) via:
`/opt/vc/bin/vcmailbox 0x00030081 40 40 0 8 0 0 0 0 0 0 0 0`
...but failed.   Then tried to overwrite the internal private key:
`/opt/vc/bin/vcmailbox 0x00038081 40 40 0 8 0xffffff 0xffffff 0xffffff 0xffffff 0xffffff 0xffffff 0xffffff 0xffffff`
...but doubt it did anything... nothing ever appeared to over-write these registers, else they're not visible to me.
being root user didn't seem to make a difference.

jdlcdl, 2023-07-05:
Older redundant/historical links than those noted above which back-fill the topic of "RPi OTP Registers" cataloged below:

https://github.com/raspberrypi/firmware/issues/974