SeedSigner/seedsigner #391: [Feature Request] Randomize or Wipe RAM bits prior to/during shutdown

bitcoinprecept, 2023-06-14:
File this under the hyper-cynical tail risk threat mitigations that is probably not worth attention in 2023, but may also be a simple enough thing that it's easy to implement by a motivated contributor.

Part of the idea and ethos of seedsigner is a stateless signing device. This relies on the commonly accepted principle that RAM loses all data retention after being powered off. Still, I've interacted with a few potential seedsigner users who are skeptical of this claim and seek more assurance that there's no possible way to read the private data held in RAM after shut down. Pointers to research or evidence that RAM is reliably unreadable after being powered down may be helpful comments here without adding explicit code to the project.

Code ideas thrown around would be to include some bit randomization inside of a secure shut down UI flow. 

@pythcoiner asks "what about running a thread who write random data (with same size than seed/words/etc) during the seed is hold in memory and only stop after we remove the seed for memory, so we can be (not absolutely but quite) sure that the seed data had been overwritten?"

@newtonick "For the Pi zero platform, it seems like the ideal way to “wipe” memory would be a kernel module that wipes all userland space as an optional shutdown procedure. In Linux it would be possible to separate userland vs kernel memory space. I don’t know now how to implement this and I’m not convinced this kind of measure has meaningful benefits."

@SeedSigner "Would it be possible to encrypt userland space in memory with a random key generated from the camera? Would eliminate the wiping time at the end of a usage session. I too am not convinced the time spent on something like this would be worth it, just spit balling."

jahangir13, 2023-06-14:
I found some pdfs to this topic:
static.usenix.org/event/sec08/tech/full_papers/halderman/halderman.pdf

www1.cs.fau.de/filepool/projects/coldboot/fares_coldboot.pdf

This one is maybe interesting. It's about encrypting the ram:
https://www.researchgate.net/profile/Peter-Peterson-3/publication/224201954_Cryptkeeper_Improving_security_with_encrypted_RAM/links/0912f50cfdf4b2e20b000000/Cryptkeeper-Improving-security-with-encrypted-RAM.pdf

pythcoiner, 2023-06-14:
something interesting [here](https://www.sjoerdlangkemper.nl/2016/06/09/clearing-memory-in-python/) for reminder:

pythcoiner, 2023-06-20:
just learn about [memoryview](https://docs.python.org/3.10/library/stdtypes.html?highlight=memoryview#memoryview) type today, maybe this can give someone idea:

```python
seed = bytearray("picture picture picture picture picture picture picture picture picture picture picture picture ", 'utf-8')

mv = memoryview(seed)

print(seed)

for i in range(20):
    mv[i] = 000

print(seed)
```

returns
```sh
bytearray(b'picture picture picture picture picture picture picture picture picture picture picture picture ')
bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00ure picture picture picture picture picture picture picture picture picture ')
```

jdlcdl, 2023-06-20:
> just learn about [memoryview](https://docs.python.org/3.10/library/stdtypes.html?highlight=memoryview#memoryview) type today, maybe this can give someone idea:

Would the idea be to store the sensitive Seed objects attributes (currently: mnemonic List[str], passphrase str, and seed_bytes bytes) as bytearray or bytes and to wipe them in the power-off/reset screens?   And then to be very careful about the copying of these elsewhere throughout the codebase?

Nice find @pythcoiner!

pythcoiner, 2023-06-20:
but  _memoryview_ can reference only bytes/bytearray type I think, so maybe we need to find a way to type casting tables/img/string ty bytearray? (I'm not even sure this is possible in python)?
Another way can be to reimplement all features having contact with the seed in C? (crazy heavy)

alltheseas, 2023-06-20:
> Part of the idea and ethos of seedsigner is a stateless signing device. This relies on the commonly accepted principle that RAM loses all data retention after being powered off. Still, I've [...]

Appreciate you opening an issue on this @bitcoinprecept.

To add context, here is the origin of the conversation: https://discord.com/channels/988346170701135902/1113605496704872529/1113879446861119528.

It is great to see multiple prospective security improving solutions being discussed, and explored. 

These all go over my technical competency, so I cannot provide useful feedback.

The specific ask is verifying the stateless claim by a cold-eyes, independent security expert. These folks are expensive, and clearly Seedsigner is FOSS with extremely limited resources.

I don't know if this possible, but I think the value and motivation that open source brings could be something the Seedsigner team can use to make a call for hardware+software+firmware security experts to volunteer expert feedback on the Seedsigner security architecture. 

Maybe such a call for help/review could be by funded by https://opensats.org/. I think Seedsigner is definitely in scope for those folks.

An independent expert could help:
1) identify unknown shortcomings, and
2) verify if the avenues discussed in this thread are addressing a real risk as identified by a security expert (and thereby judge if worth pursuing), and
3) help the PM and team prioritize various security improvements.

Thanks again for all of yalls work. I'm a yuge fan, and I hope this discussion helps improve the product in the long-term 🙏 .