SeedSigner/seedsigner #393: [Feature] Seed XOR

pythcoiner, 2023-06-18:
Does someone already plan to implement a Seed XOR feature on seedsigner?

jdlcdl, 2023-06-18:
I had planned to implement SeedXOR [here](https://github.com/jdlcdl/seedsigner/tree/seed_xor), in a less than ideal flow where you can XOR a loaded seed against a pending seed before finalizing.  @dipunm suggested a more ideal flow where pending seeds are loaded and XORed against each other before the final one is loaded, but I never did that work.

There have been plenty of discussions around how it could be misused/misunderstood creating problems that are avoided using multisig.  An old issue with discussions is [here](https://github.com/SeedSigner/seedsigner/issues/43)

pythcoiner, 2023-06-19:
Nice work, I'll dig in and comment there

closed 2023-06-19 by pythcoiner