SeedSigner/seedsigner #396: [Enhancement] Increase dice roll entropy capture to 100 rolls

bitcoinprecept, 2023-06-26:
Today, Seedsigner allows for 99 dice rolls to capture entropy during seed creation. The requested enhancement is to increase that number to 100, not because of any meaningful increase in entropy but rather to align with how other tools in the ecosystem are built such that users can have increased verification across devices when creating keys by dice roll. Important caveat - it should still be confirmed that there is a uniform method being used (see note at bottom about iancoleman's tool).

The topic was initiated by Twitter user @svrgnty's [hardware wallet reviews](https://twitter.com/svrgnty/status/1672952628567851008?s=20). [Here](https://twitter.com/KeithMukai/status/1672997886752088067?s=20), @kdmukai explains the rationale for seedsigner currently being designed with only 99 rolls:

"Rationale for the 99-role limit:

2^256 = 1.15e77
6^99 = 1.08e77

6^100 way overshoots: 6.53e77

Additional rolls aren't adding any real benefit (it all gets sha256'ed back down to 256 bits anyway).

Worse: If the UI invites you to roll more, a naive user will think...'oh, more is always better, so let's do 200 rolls to be SUPER-secure!'

Not only would it be wasting their time, it would create a false sense of extra security for them and potentially put doubts into people who didn't do extra rolls.

So I view it as a lose-lose."

Svrgnty replied "Yeah, understand, not arguing for 200 roles, only mentioned it because most people/guides recommend 100 rolls, and if you are inputting the rolls on multiple devices (e.g. specter or coldcard), you may input 100 rolls, then realize seedsigner goes to 99, needing to start over"

Keith replied "That's the best argument I've heard for 100 rolls.

When I was doing the research for number of rolls, there were tons of old discussions that helped me land on 99. But ultimately 99 vs 100 doesn't really matter for any given seed.

But for external validation, ya...**Though that being said, we'd have to see if everyone is using the same approach for how to process those 100 rolls.

[http://iancoleman.io](https://iancoleman.io/) is a master ref tool, but his dice method doesn't match how we or Coldcard or probably/hopefully most others calculate it (bits vs chars)**.

Anyway, If everyone else is at 100, we should probably join the party.

I think it's unnecessary, but one more roll won't hurt anyone.

Marc-Gee, 2023-06-27:
Thx @jdlcdl ! 
Agreed with you checking the 'references of others '. Thx for checking that back. Discrepancy was also noted by me. 

50 rolls for a 12 word and 100 rolls for a 24 word , does help it to be slightly simpler, and doesn't hurt. 

Getting secure is hard enough, let's not flood people's brains too much. 99 is fine for entropy, but  a round 50 or 100 is easier to 'stick'. 

To the point of more security incorrectly being assumed by >100 rolls, we should set 50 and 100 as the respective maximums (and minimums). 
Ie you need 50 or 100 rolls (of a good quality 6-sided dice). 

Marc.

manny4848, 2026-10-18:
Yes please! Please remove the hard cap limit for dice rolls but as mentioned earlier you could send a message at 50 and 99/100 that it is good enough. I would really like it for offline cross device verifying seed generation for dice rolls.