SeedSigner/seedsigner #398: A mainnet seedsigner will sign a PSBT created for testnet3, displaying mainnet addresses.

jdlcdl, 2023-07-13:
Posting this for informational purposes only.  I'm not convinced that anything "needs" be changed.  Can close this after discussions so it will still be searchable in the future.

I believe this has been reported in the past, but before today I was not able to read-enough into the complaint to be able to reproduce it.

In telegram, it was reported:

```
Daniel, [7/13/23 2:52 AM]
Hello! We were yesterday testing some multisig in our local meetup and we found something awkward. I don’t know if it’s a bug or intended behavior: 

Created multisig 2-3 in sparrow and nunchuk, testnet. I’m some moment the power was off on the seed signer and it reseted on mainnet, we restore the 3 signatures and make a transaction. Then we signed those transactions with the seedsigner in mainnet, the address shown on the seedsigner were different than the one we have on sparrow/nunchuk. The transaction was equally signed and broadcasted without issues.

Daniel, [7/13/23 2:53 AM]
The only problem is the screen on the seedsigner showing a different address than sparrow. Also tried the seedsigner in regtest and signed transactions in testnet… this is intended behavior?

Daniel, [7/13/23 2:53 AM]
Is there a way to alert or know if the transaction is on a different network? Thank you! 🙏
```

Indeed, I was able to reproduce this by setting seedsigner to "mainnet" and creating similar 2of2 PSBT transactions from Specter Desktop and Sparrow.

Seedsigner loads the PSBT, does not warn, displays mainnet addresses for the correct derivation path and address index, correctly identifies change similarly for a mainnet address, and signs the transaction.

I've concluded that at this time, seedsigner did nothing wrong because there is no information in the PSBT (except for perhaps hinting that the derivation path is "non-standard" while it would still be perfectly valid) that indicates the transaction is for one network or another.  The resulting signed transaction would appear to be of valid construction for any network, and would only fail once the node realizes that the inputs are not valid unspent utxos... which is out of reach for any airgapped signing device.  It is up to the user to review the payment/self-transfer and change addresses ON THE SEEDSIGNER to know that something is wrong.  But if they didn't, since the utxo being spent do not exist on the wrong network, the resulting signed PSBT could not be broadcasted, and no harm would be done.

Parsing a Sparrow PSBT for a 2of2 Native Segwit wallet on testnet3
-------------------------------------------------------------------------------------------------

A PSBT can be thought of a string of bytes representing (magic, global-map, input-map, output-map).

Nothing I saw "screams" that this transaction is for one network or another, but there was a hint in derivation paths on inputs and outputs.

## magic
"psbt" + 0xff, just as expected

## global-map
PSBT_GLOBAL_UNSIGNED_TX containing a rawtransaction w/o signature/witness data.

0x00 just as expected to terminate the global-map

## input-map
PSBT_IN_NON_WITNESS_UTXO the complete segwit rawtransaction previously confirmed in the blockchain -- the only input being spent.

PSBT_IN_WITNESS_UTXO amount and scriptPubKey of the input.

PSBT_IN_WITNESS_SCRIPT the witnessScript for the input

2 PSBT_IN_BIP32_DERIVATION values, one for each pubkey that must sign, w/ master fingerprint and derivation path of the input being spent
**this might be useful if we'll hint that "maybe on wrong network" or "non-standard path".**

0x00 just as expected to terminate the input-map

## output-map
PSBT_OUT_WITNESS_SCRIPT the ouput witnessScript for the payment leaving the wallet.

2 PSBT_OUT_BIP32_DERIVATION values, one for each pubkey in the wallet, w/ master fingerprint and derivation path of the change coming back.
**this might be useful if we'll hint that "maybe on wrong network" or "non-standard path".**

0x00 as expected to terminate the output-map

0x00 as expected to terminate the psbt.

jdlcdl, 2023-07-14:
All of that manual parsing yesterday was for a SpecterDesktop PSBT.  My statements above remain my opinion for SpecterDesktop PSBTs... but since realizing that I had not addressed the PSBT from the original complaint, I spent time implementing a half-finished bip174 psbt parser, that will make looking at psbts much easier in the future.

SPARROW DOES INDEED HAVE ENOUGH INFO, via an optional field, for seedsigner to improve/warn the user that they're on the wrong network.

The PSBT global-map has, optionally since version 0, keytype:PSBT_GLOBAL_XPUB, which contains keydata that is a complete 78 byte bip32 xpub where the first 4 bytes are the network magic for mainnet or testnet.  The value is less interesting, it's the master-wallet-fingerprint + 4byte little endian path elements, concatenated.

I will need to take a look at our psbt parser and see where we might make changes to take advantage of this extra field that exists in Sparrow, but not SpecterDesktop xpubs.  As well, worth looking at similar from our other supported coordinators so that when a change is made, we'll know that it helps users of x,y,z coordinators.

kdmukai, 2023-07-14:
@jdlcdl it may also make sense to make PRs into `embit` and Specter Desktop or at least see if there have been any discussion of that field in those repos.

jdlcdl, 2023-07-14:
I've since learned that nunchuk also has the PSBT_GLOBAL_XPUB present in at least single-sig native segwit PSBTs.

Also that at a glance, embit seems to faithfully support bip174 (either outright else PSBT objects will have inputs.unknown and outputs.unknown so that unsupported keytypes will get passed through.)  For that which is fully supported by embit, which includes PSBT_GLOBAL_XPUB, can do it the embit friendly way via PSBT.xpubs... but I'll have to verify/play.

jdlcdl, 2023-07-14:
I've poked around in embit and have a sample script that I believe illustrates PSBT_GLOBAL_XPUB can be accessed the 'embit' way via embit.PSBT.xpubs which behaves like a dict whose key is an embit.HDKey, which will have a ".version" attribute set to the network (main or test) that the wallet was created for (signet and regtest are same as test).

```python
from base64 import b64decode
from embit.psbt import PSBT
from embit.networks import NETWORKS

if __name__ == '__main__':
    

    psbts = {
        'main': 'cHNidP8BAHECAAAAAdCEXX0XBXxR6G0dD95gpJhHbIT/lF60ThGfrg7my8AIAQAAAAD9////AhAnAAAAAAAAFgAUrqizVmmVhneVbnOSCadUp+hpTEQxANwAAAAAABYAFJNP3T/N7XdJnCTEsdK4zTNlr022AAAAAE8BBIiyHgMMRg9ggAAAABg5cK8myN4on9odN8WutS/eUTTRSGzSr0ROX6xjaeM/Az24ab/atYzPxNKwag3Sjowqk0rf5frw7z1X+mDIjeglEA55sJ1UAACAAQAAgAAAAIAAAQByAgAAAAHLKVpQmEIPT7rcZN1Ep/ucMTCGJ88GJb2gibOBJsdHCQAAAAAA/f///wJAS0wAAAAAABepFBfnfQKWZZ5DrrBD5SXD2zb5Ugkih84n3AAAAAAAFgAUJtD2n29RhvcJR+meZzz4H6dqC/whlRUAAQEfzifcAAAAAAAWABQm0Pafb1GG9wlH6Z5nPPgfp2oL/CIGAlc/HJvHl4CaJdIDHFJw8JLNavfO6HD+2o/UN82xoydAGA55sJ1UAACAAQAAgAAAAIABAAAAAwAAAAAiAgPj1ZMkmMtFkxy9dA5V8diy7EQBFw1njqZIOwt7+9wGLhgOebCdVAAAgAEAAIAAAACAAAAAAAkAAAAAIgICi586gJLYLYS2nojGdgII8F/lCij39uaO8p03Ihyng8EYDnmwnVQAAIABAACAAAAAgAEAAAAGAAAAAA==',
        'test': 'cHNidP8BAHECAAAAAdCEXX0XBXxR6G0dD95gpJhHbIT/lF60ThGfrg7my8AIAQAAAAD9////AhAnAAAAAAAAFgAUrqizVmmVhneVbnOSCadUp+hpTEQxANwAAAAAABYAFJNP3T/N7XdJnCTEsdK4zTNlr022AAAAAE8BBDWHzwMMRg9ggAAAABg5cK8myN4on9odN8WutS/eUTTRSGzSr0ROX6xjaeM/Az24ab/atYzPxNKwag3Sjowqk0rf5frw7z1X+mDIjeglEA55sJ1UAACAAQAAgAAAAIAAAQByAgAAAAHLKVpQmEIPT7rcZN1Ep/ucMTCGJ88GJb2gibOBJsdHCQAAAAAA/f///wJAS0wAAAAAABepFBfnfQKWZZ5DrrBD5SXD2zb5Ugkih84n3AAAAAAAFgAUJtD2n29RhvcJR+meZzz4H6dqC/whlRUAAQEfzifcAAAAAAAWABQm0Pafb1GG9wlH6Z5nPPgfp2oL/CIGAlc/HJvHl4CaJdIDHFJw8JLNavfO6HD+2o/UN82xoydAGA55sJ1UAACAAQAAgAAAAIABAAAAAwAAAAAiAgPj1ZMkmMtFkxy9dA5V8diy7EQBFw1njqZIOwt7+9wGLhgOebCdVAAAgAEAAIAAAACAAAAAAAkAAAAAIgICi586gJLYLYS2nojGdgII8F/lCij39uaO8p03Ihyng8EYDnmwnVQAAIABAACAAAAAgAEAAAAGAAAAAA==',
    }

    for my_net in ['main', 'test', 'regtest', 'signet']:
        print("\nSigner intends to use network: %s" % my_net)

        for psbt_net, psbt in psbts.items():
            psbt = PSBT.parse(b64decode(psbt))

            xpub_versions = set([k.version for k in psbt.xpubs])
            if NETWORKS[my_net]['xpub'] in xpub_versions:
                print('processing a %s psbt while signer on network:%s...' % (psbt_net, my_net))
            else:
                print('WARNING: psbt network:%s while signer on %s!' % (psbt_net, my_net))
```

jdlcdl, 2023-07-15:
> @jdlcdl it may also make sense to make PRs into `embit` and Specter Desktop or at least see if there have been any discussion of that field in those repos.

On SpecterDesktop, I have changed my device from "seedsigner" to "coldcard" and exported a PSBT which does contain the desired PSBT_GLOBAL_XPUB keytype.   As we get closer to merging support for this in seedsigner, I think that the solution will be to request a configuration change for seedsigner signing devices so that the optional param "xpubs=True" is passed to the Wallet.fill_psbt() method.  However, I don't suspect that there would be any problem if SpecterDesktop did this before we supported it... we're simply ignoring it even though embit has parsed this info and made it easily available.

https://github.com/cryptoadvance/specter-desktop/blob/8f9c95fe1b45e638d2997e1392857a96b391f09b/src/cryptoadvance/specter/wallet/wallet.py#L1756

I will refrain from adding a new issue to the cryptoadvance/specter-desktop repo until others have taken a look at its resolving pr.  Once others are ok, perhaps I'd word it as follows:

---

Feature Request: PSBT_GLOBAL_XPUB for PSBTs exported to SeedSigner
---------------------------------------------------------------------------------------------------------

PSBT_GLOBAL_XPUB is not an included keytype for PSBTs exported to seedsigner; it is available when device is "coldcard"

While seedsigner does not yet take advantage of this field, it would be available via embit in PSBT.xpubs and would be useful to inform the user when they have setup seedsigner to use testnet but are about to sign a PSBT intended for mainnet.  Without this information, seedsigner processes the PSBT all the way to a valid signature but shows testnet addresses instead of mainnet addresses.