SeedSigner/seedsigner #419: [Feature Request] Clear Passphrase / Change Passphrase

bitcoinprecept, 2023-08-01:
This feature request is basically to bring level the feature set offered by Foundation Passport, which I would find useful on Seedsigner and suspect is not a huge lift. I'm not sure if the persistent memory of the Passport vs stateless nature of the Seedsigner plays a role in whether this is possible, but perhaps that's a constraint.

The desired feature is to be able to clear a passphrase that you've previously entered and return to the "bare" seed. I feel like this should be possible since at any time you can backup your seed to SeedQR even if a passphrase is loaded on top. Then, if that is possible, it should also be possible to simply change passphrase (or load additional passphrase wallet, depending on you want GUI to flow).

I'm sure there's some GUI elements to think through if implementing this. Thanks as always for your consideration.

jdlcdl, 2023-08-02:
> The desired feature is to be able to clear a passphrase that you've previously entered and return to the "bare" seed. I feel like this should be possible since at any time you can backup your seed to SeedQR even if a passphrase is loaded on top.

This understanding is correct.  As well, gui-gotchas would have to be worked out.  Above all, IF it's a feature that is truly desired enough to make it all the way to 'merged'; it makes sense to have that confirmation before too much effort is spent on research/implementation.

As soon as we get to 0.7.0 release, I'd be willing to look into this with you, but before then, I'm more concentrated on the PRs that are close to the finish line -- which can be pushed across before it happens.  As 0.7.0 is released, I'd like to have a few weeks of complaint-response time freed-up in advance to deal with "we forgot to update these docs" fixes so they can quickly be merged (since users read the latest repo anyways) where users will have full access to understand and take advantage of the new benefits just released.

jdlcdl, 2023-12-30:
A strong opinion in support of this functionality was expressed in the telegram group within the past 7 hours.

G0RR4, 2024-01-03:
Most people I know use passphrases to manage multiple wallets. It is very **UNPRACTICAL** on seedsigner that you have **to re-enter** your 24 base seeds in order to switch from one to another wallet. Add / Remove Passphrase functions are essential for cold wallets.

jdlcdl, 2024-02-22:
> A strong opinion in support of this functionality was expressed in the telegram group within the past 7 hours.

ditto

jdlcdl, 2024-08-03:
Interest in this feature was repeated in the telegram group today.

alvroble, 2024-08-16:
@kdmukai @newtonick, what do you think about adding this feature to clear or edit the passphrase of an existing seed?

I could take a look at it and implement a UI flow to make this possible. One design decision to consider would be choosing between these two options:

- Edit the existing seed in storage.
- Create a new seed with the modifications, keeping both the "old" and "new" seeds in storage.

Maybe, both could be implemented and users could choose the option that best suits their needs.

kdmukai, 2025-06-14:
A bit of background for everyone (okay, this turned into a deeper SeedSigner history lesson, but worth documenting somewhere):

The earliest versions of SeedSigner used a DOS-like UX consisting of rows of text selections.

[image]

The UX in this era had been built up piecemeal: the first early prototype steadily accumulated more and more features. Naturally not every new piece had a proper fit, but as is often the case, it's easier to stuff a new feature in a slightly awkward place than it is to make broader reorganization changes.

The v0.5.0 release was the first to incorporate @easyuxd's modern UX design. But v0.5.0 wasn't just putting a pretty skin over the existing functionality and user flows. We took the opportunity to take a fresh look at ALL of the flows. It was no longer acceptable to have those piecemeal features stuffed wherever they sorta-kinda-but-not-really fit. We were committed to defining the rigid paths and sub-loops that made the most sense.

[image]

It's very hard now to convey just how hard this task was at the time. I think we absolutely nailed it; those decisions feel so intuitive now that those earlier decisions seem invisible or inconsequential: of course it works this way because how else could it?! Except we did agonize over them, rearranging and reorganizing until everything fit just right. Thus my moveable paper card approach to facilitate easy rearranging.

[image]

Now, notice the Advanced Seed Options card: "Add Passphrase" is right there!

IF you understand what a bip39 passphrase does, it's obvious why that button is there. And obvious when you should ignore it.

So a few thoughts:

1. It's annoying that it's the first option in this list. If I don't need to enter a bip39 passphrase or if I never use them, then it's a persistent obstacle to navigate past.
 
2. But if I DO need to apply a bip39 passphrase, you want its position to imply the necessary order of operations: apply the passphrase BEFORE you export the xpub!! So it SHOULD be high up on the list.

3. Here was the biggest problem: What if you had no clue what a bip39 passphrase is? And note in this prototype, we don't even say "bip39". It's just "Add Passphrase". The biggest fear with bip39 passphrases (notice how I always write that full phrase "bip39 passphrase" because even in discussions it gets confusing if you aren't super specific) is that a naive user misunderstands what it is and inadvertently exports an xpub for the altered version of their seed. They mistakenly think it's about storage security without realizing they have created a whole different wallet universe for their funds; "oh well, if I lock myself out of this seed no big deal, I still have my mnemonic backup so I'm all good!" 😭

"Add Passphrase" = "oh, passphrases are better security, I should add that" = "how did I lose access to my funds?!!!"

Yes, these concerns are a bit exaggerated and, no, we can't prevent ignorant people from making all possible mistakes.

But the other issue is that our screen is just so small that we didn't think we could adequately warn or educate ignorant users about exactly what they were about to do.

And if "Add Passphrase" is constantly staring them in the face every time they interact with their seed... it just seemed like too many footguns waiting to happen.

This feature was the HARDEST to get right. We agonized over it.

So notice what our current solution is:

[image]

We intentionally don't use the verb "Add". The hope is that "BIP-39 Passphrase" is just technical sounding enough that it would be intimidating to a new user. They are subtly DISCOURAGED from clicking on it rather than constantly tempted to "Add" extra security.

In fact, we even break from Easy's usual UX rules: "Done" is the FIRST option in the list. We do everything we can to discourage users from wandering into footgun land.

And once the loaded seed is "finalized", the user is never presented with an option to add/remove the bip39 passphrase.

But why is that? Because then we'd be right back to "Add Passphrase" (or more likely: "Add BIP-39 Passphrase" or, I suppose, the same intentionally cryptic "BIP-39 Passphrase") in the Seed Options menu.

But what if the user does need to change their bip39 passphrase?

The bip39 passphrase entry flow is written to allow for a review-and-edit loop as needed.

[image]

It's on the user to verify that the resulting new fingerprint matches what they were expecting. THIS is your chance to get it right.

Obviously as described in other comments here, we don't allow the use case where a user wants to intentionally re-use the same seed with a different bip39 passphrase.

But given all the juggling tradeoffs discussed in detail above, I think you can at least appreciate our perspective: yes, I can see the utility of this additional use case, but at what UX cost? Given all the effort we went through to mitigate worst case footgun possibilities, adding back "Add" somewhat undoes everything.

## Alternate flows
We actually have a few configurable alternate flows:

[image]

The default ("enabled") was even a difficult decision. My strong preference based on everything above was "disabled" (in which case a newly loaded seed is auto-finalized and never even presented with the option). My thinking was that a user advanced enough to know what a bip39 passphrase is can deal with going into the advanced settings and enabling it.

[... cut ...] tive.
* Or maybe the bip39 passphrased variants are treated like child seeds in the UX? I like that better but is creating a larger departure from how in-memory seeds (including those with bip39 passphrases) are currently managed.
* Perhaps a hybrid approach borrowed from some bip85 implementation ideas (I think there's a PR or an Issue that suggests an icon in the list of in-memory seeds that displays the hierarchical relationship:

  ```
  * Seed A fingerprint
  * ⮑ Child 1
  * ⮑ Child n
  ```