SeedSigner/seedsigner #422: frank-s01-v1.0 crashes

matheusdiogenesandrade, 2023-08-04:
Hello.

Recently, I obtained a fresh DIY seedsigner kit. After embedding it and turn it on, I tried to use the camera. However, two seconds after the camera appears, seedsigner crashes, to the point no command from the buttons work, i.e. doesn't matter what I press, nothing happens. I tried to reinstall the seedsigner software again in the SD card. But no result. Then I tried to turn it on out of the seedsigner box (that orange one), but no results as well. I would like to know if some has faced such a issue before.

Thanks and best regards.

jdlcdl, 2023-08-04:
Camera issues are frequently related to (in order):
* a power supply that is not sufficient, try another with a higher output-amps rating.
* a bad camera ribbon connection at the raspberry pi board.  Disconnecting that ribbon and reconnecting it, even if it appears like it's seated properly, fixes many problems.
* the ribbon can be inserted upside-down without it being obvious that this is incorrect... it fits perfectly the right way and the wrong way.  The electric connectors on the ribbon must be facing down against the pi0 board when inserted into its connector.

matheusdiogenesandrade, 2023-08-04:
> Camera issues are frequently related to (in order):
>
> * a power supply that is not sufficient, try another with a higher output-amps rating.
> * a bad camera ribbon connection at the raspberry pi [...]

Thanks for the attention, but the issues still persist. Updates in the initial issue description.

newtonick, 2023-08-04:
I've seen this issue with power supply issues in the past. Have you tried running an older version of seedsigner (like release 0.5.1)? https://github.com/SeedSigner/seedsigner/releases/tag/0.5.1

If the power supply is not the issue, give the older OS a try to help diagnosis if something software related is influencing the crash/freeze your experiencing.

closed 2023-08-04 by matheusdiogenesandrade