SeedSigner/seedsigner #426: [Feature Request] Load BIP-85 child seeds directly into memory after calculating from parent seed

bitcoinprecept, 2023-08-08:
The current BIP-85 implementation on Seedsigner allows you to calculate child seeds, but it does not allow you to directly load them into memory to interact with them for signing or XPUB export etc.

Would it be possible to offer the user the option to directly load it into memory after calculation from parent? This would allow extra opsec of not even having to document the child seed seedphrases or QR codes anywhere.

I understand the current GUI is designed to discourage use of BIP-85, but I personally would like to use the feature in this way. I suspect others might as well.

jdlcdl, 2023-08-08:
Issue to #354

bitcoinprecept, 2023-08-08:
Ah thank you, @jdlcdl .

@newtonick this issue can be closed to defer to the pre-existing one. Normally I review and catch duplicates before submitting - apologies.

closed 2023-08-08 by bitcoinprecept