SeedSigner/seedsigner #429: System Error when attempting to export Taproot xpub for Multisig

thebrandonlucas, 2023-08-13:
To reproduce:
I added "Taproot" to the list of possible address types in settings.

Then, I went to export my xpub for multisig wallet by doing: 
- Seeds -> Select Seed -> Export Xpub -> Multisig -> Taproot -> Choose wallet -> "I understand" (on privacy leak error).

The next screen shows the following "System Error":
`embit_utils.py, 43, in get_standard_derivation_path.py. Taproot multisig/musig not yet supported`
[image]

As stated in [whentaproot.com](https://whentaproot.org/#features), there are many privacy benefits (among other things) with Taproot, particularly with obfuscating multisig/single-sig outputs in this case. Please let me know if I can help or provide any other context with this issue!

newtonick, 2023-08-13:
Which coordinator are you using for multisig taproot? I don’t know of any QR supported coordinators that support anything but p2tr, which is single sig.

kdmukai, 2023-08-13:
@thebrandonlucas forgive me if you're already deep w/this knowledge, but unless you really know what you're doing, "multisig" + taproot probably isn't what you think it is.

Rabbitholes to dive into:
* Current multisig opcodes vs new taproot multisig opcodes
* "Musig" != "multisig"
* FROST

We did this panel over a year ago and, afaik, we're not really any closer to real-world, pleb-ready taproot multisig implementations:
https://www.youtube.com/watch?v=0tbi2kYJiU0

kdmukai, 2023-08-13:
That being said, @newtonick, we should change this error to be something like a `NotYetImplemented` exception. We should reserve System Error for actual code crashes.

thebrandonlucas, 2023-08-13:
> @thebrandonlucas forgive me if you're already deep w/this knowledge, but unless you really know what you're doing, "multisig" + taproot probably isn't what you think it is.
>
> Rabbitholes to dive [...]

There's definitely a lot for me to better grasp here -- I appreciate the suggestions!