SeedSigner/seedsigner #440: [Feature Request] Scramble keyboard starting position to mitigate microphone listening attacks

bitcoinprecept, 2023-08-14:
Given the recent uptick in stories about microphone based listening attacks on keyboard strokes improving due to AI, one community member (Johny D) requested a feature to scramble the seedsigner's keyboard starting position after every input, thus making it more difficult to analyze likely inputs based solely on ambient sound of device clicks.

jdlcdl, 2023-08-14:
I just recently opined "Im not sure the words `import random` belong anywhere in our codebase." and here's an example of where it probably does.

So as I understand it, instead of leaving the cursor at the last character where a 'press' occurred, randomize where it will be on the next word? or would it move around after every character typed into each word?  I hope that the key positions would not be scrambled.

bitcoinprecept, 2023-08-14:
I don't think it makes sense to randomize key positions on our keyboard. That would break people's brains. 

But when entering seed words, I'd suggest at the start of each new word, the cursor randomly starts in a new location on the keyboard. 

It's probably not doable when entering passphrases because when would you randomly jump? Maybe every X characters if user was informed it would be happening as to not be confused.

kdmukai, 2023-08-14:
My understanding of that attack was that AI was trained to listen to typing on a specific, known physical keyboard (e.g. specific model of a Macbook Pro) to learn all the acoustic nuances of each key stroke, chassis interactions/resonances, etc.

An AI could be trained in a similar way to study the SeedSigner input mechanism. On the one hand, a single joystick click per key move would be more easily solvable (mathematically), but on the other hand, the ability to press and hold -- combined with our observed different fps rates on different boards and on different OSes -- creates complications for sussing out the keys selected. Different enclosure designs might create more variables. Different users also take more or less advantage of the "Pac-Man" looping off the far edges to get to their next key.

I would file this under: Possible but not probable any time soon.