SeedSigner/seedsigner #449: Bug Report: "Select Diff Seed" Unexpected Behavior

SeedSigner, 2023-08-22:
Background: 2-of-3 wallet with two of the co-signers already active on the SeedSigner. I signed with one of the good co-signers, then attempted to sign with that co-signer again. I received the error "PSBT Error" "Signing with this seed did not add a valid signature" and was prompted to "Select Diff Seed".

When I selected this option, I was returned to the beginning of the transaction review workflow rather than to a list of currently-loaded seeds, which would seem to be the expected behavior.

jdlcdl, 2023-08-22:
I DONT HAVE MUCH TO SHARE, just my thoughts... which are far from an answer.

I suspect that you ARE being redirected back to [PSBTSelectSeedView](https://github.com/SeedSigner/seedsigner/blob/beaf4ae3a465d7133f063ca4661c8f46452ac4fe/src/seedsigner/views/psbt_views.py#L30) but that you're immediately redirected into the PSBTOverviewView because self.controller.psbt_seed is still set from your last iteration.

Admittedly, I'm not familiar with this code.  I don't know if the PSBTSigningErrorView could unset the psbt_seed so that a redirect won't happen, or if the check below the line liked above should be checking for one-matching-seed exactly, and failing if there are 2 matching seeds, so that you can have a choice to choose just one... or if it should be checking for a matching seed that hasn't already signed the psbt.

jdlcdl, 2023-08-22:
Im having a hard time duplicating.  Is it correct to:
* load threshold seeds into seedsigner,
* scan psbt via MainMenu / Scan,
* sign with first seed,
* scan psbt w/1 sig via MainMenu / Scan,
* let it choose the first seed again and try to scan?

I'm not getting to the point where "Signing with this seed did not add a valid signature" (or rather what I think it means, did not add an additional signature).  Instead, it just gives me the qrcode that I scan back into Specter Desktop... which realizes I still need other keys to sign.

SeedSigner, 2023-08-22:
As far as I can tell you have the sequence correct. To arrive at this I...

- loaded two of the three required keys into signer
- scan proposed spend into signer
- sign with first seed & send PSBT back into coordinator
- scan proposed spend into signer again
- attempt to sign with the same seed again

Not sure that it would matter but I am using Sparrow to produce this error screen -- maybe Sparrow is including the completed signature into what is passed back to the signer the second time?

jdlcdl, 2023-08-22:
>  I am using Sparrow to produce this error screen

That was it, I confirm the error.

closed 2023-08-24 by newtonick