SeedSigner/seedsigner #450: Creating seed via image entropy w/ another seed loaded raises exception

jdlcdl, 2023-08-23:
As of commit beaf4ae which we've been testing for 1 week, I've experienced an exception while one seed was already loaded and trying to create a new seed from image entropy.  It's a DireWarning like:

System Error
ValueError
"Image.py 1505 in paste", "cannot determine region size; use 4 item box"

I've just updated my testing head for Nick's recent merges, to commit 8b5c8af5, and the above error remains.

Image.py is part of PIL, so it's a hint, but not so helpful.  A backtrace from a dev pi w/ console output looks like:

```
ValueError: cannot determine region size; use 4-item box
cannot determine region size; use 4-item box
Traceback (most recent call last):
  File "/home/pi/seedsigner-dev/src/seedsigner/controller.py", line 307, in start
    next_destination = next_destination.run()
  File "/home/pi/seedsigner-dev/src/seedsigner/views/view.py", line 161, in run
    return self._run_view()
  File "/home/pi/seedsigner-dev/src/seedsigner/views/view.py", line 156, in _run_view
    return self.view.run()
  File "/home/pi/seedsigner-dev/src/seedsigner/views/tools_views.py", line 71, in run
    ret = ToolsImageEntropyLivePreviewScreen().display()
  File "/home/pi/seedsigner-dev/src/seedsigner/gui/screens/screen.py", line 59, in display
    raise e
  File "/home/pi/seedsigner-dev/src/seedsigner/gui/screens/screen.py", line 56, in display
    return self._run()
  File "/home/pi/seedsigner-dev/src/seedsigner/gui/screens/tools_screens.py", line 68, in _run
    self.renderer.canvas.paste(frame)
  File "/home/pi/.envs/seedsigner-dev-env/lib/python3.10/site-packages/PIL/Image.py", line 1505, in paste
    raise ValueError("cannot determine region size; use 4-item box")
ValueError: cannot determine region size; use 4-item box
```

jdlcdl, 2023-08-23:
This morning, Nick suggested changing "gpu_mem" from 64 to 128 in config.txt on the pi0.
This worked for me on pi0, I can no longer reproduce this error.

However, on Raspi-OS I have always used only 16mb for gpu_mem.  Changing to 128 and even 256 does NOT resolve the error.  I still believe that we should track down exactly where this became a problem.  I suspect it's recent, but I'm not sure, It's not a regular habit of mine to load more than one seed at a time... but I've certainly done this in the past and never seen this error before.

kdmukai, 2023-08-23:
I ran into that maybe a month ago but at the time it was because that `paste` operation was trying to paste `None` instead of image data. It was just a bug in my in-progress code at the time that I resolved. No clue yet what might be causing it this time.

PR #453 (2023-08-24): [Bugfix] Entering image entropy via long click raises PIL Exception
fixes #450

## The problem
Prior to this PR, the `frame is None` check did not affect later attempts to paste that frame into the Renderer upon final image click.

Note: When the camera first starts up, it can initially return `frame == None` for a few cycles.

If the user enters image entropy (Tools -> New seed) via a long click (click-and-hold), they enter a race condition with the camera: will the camera return a valid `frame` before the long click repeat triggers the final image logic?

## Resolution
This PR:
* reorganizes [... cut]
commit 91b02061 (2023-08-24): bugfix for entering image entropy via long click

fixes #450
closed 2023-08-24 by newtonick