SeedSigner/seedsigner #456: Message signing error handling

hax0rbana-adam, 2023-08-26:
# Problem
When a user scans a raw message to be signed (not wrapped in Specter Desktop QR format), SeedSigner will produce two errors:

1. Tells the user that the QR code is an unknown type
2. System error caused by a KeyError exception

I have verified this at commit b9eee39 of the `dev` branch. All line numbers refer to that point in time.

[image]

[image]

<details><summary>Stacktrace</summary>
<p>

```
Aug 26 02:23:39 raspberrypi python3[842]: Traceback (most recent call last):
Aug 26 02:23:39 raspberrypi python3[842]:   File "/home/pi/seedsigner/src/seedsigner/controller.py", line 307, in start
Aug 26 02:23:39 raspberrypi python3[842]:     next_destination = next_destination.run()
Aug 26 02:23:39 raspberrypi python3[842]:   File "/home/pi/seedsigner/src/seedsigner/views/view.py", line 161, in run
Aug 26 02:23:39 raspberrypi python3[842]:     return self._run_view()
Aug 26 02:23:39 raspberrypi python3[842]:   File "/home/pi/seedsigner/src/seedsigner/views/view.py", line 156, in _run_view
Aug 26 02:23:39 raspberrypi python3[842]:     return self.view.run()
Aug 26 02:23:39 raspberrypi python3[842]:   File "/home/pi/seedsigner/src/seedsigner/views/seed_views.py", line 1975, in run
Aug 26 02:23:39 raspberrypi python3[842]:     selected_menu_num = self.run_screen(
Aug 26 02:23:39 raspberrypi python3[842]:   File "/home/pi/seedsigner/src/seedsigner/views/view.py", line 109, in run_screen
Aug 26 02:23:39 raspberrypi python3[842]:     self.screen = Screen_cls(**kwargs)
Aug 26 02:23:39 raspberrypi python3[842]:   File "<string>", line 21, in __init__
Aug 26 02:23:39 raspberrypi python3[842]:   File "/home/pi/seedsigner/src/seedsigner/gui/screens/seed_screens.py", line 1516, in __post_in
Aug 26 02:23:39 raspberrypi python3[842]:     text=self.sign_message_data["message"],
Aug 26 02:23:39 raspberrypi python3[842]: KeyError: 'message'
Aug 26 02:23:39 raspberrypi python3[842]: 'message'
Aug 26 02:23:39 raspberrypi python3[842]: Traceback (most recent call last):
Aug 26 02:23:39 raspberrypi python3[842]:   File "/home/pi/seedsigner/src/seedsigner/controller.py", line 307, in start
Aug 26 02:23:39 raspberrypi python3[842]:     next_destination = next_destination.run()
Aug 26 02:23:39 raspberrypi python3[842]:   File "/home/pi/seedsigner/src/seedsigner/views/view.py", line 161, in run
Aug 26 02:23:39 raspberrypi python3[842]:     return self._run_view()
Aug 26 02:23:39 raspberrypi python3[842]:   File "/home/pi/seedsigner/src/seedsigner/views/view.py", line 156, in _run_view
Aug 26 02:23:39 raspberrypi python3[842]:     return self.view.run()
Aug 26 02:23:39 raspberrypi python3[842]:   File "/home/pi/seedsigner/src/seedsigner/views/seed_views.py", line 1975, in run
Aug 26 02:23:39 raspberrypi python3[842]:     selected_menu_num = self.run_screen(
Aug 26 02:23:39 raspberrypi python3[842]:   File "/home/pi/seedsigner/src/seedsigner/views/view.py", line 109, in run_screen
Aug 26 02:23:39 raspberrypi python3[842]:     self.screen = Screen_cls(**kwargs)
Aug 26 02:23:39 raspberrypi python3[842]:   File "<string>", line 21, in __init__
Aug 26 02:23:39 raspberrypi python3[842]:   File "/home/pi/seedsigner/src/seedsigner/gui/screens/seed_screens.py", line 1516, in __post_in
Aug 26 02:23:39 raspberrypi python3[842]:     text=self.sign_message_data["message"],
Aug 26 02:23:39 raspberrypi python3[842]: KeyError: 'message'
```

</p>
</details> 

# Steps to reproduce
1. Import or create a key
2. Enable message signing in the options
3. Seeds -> Key ID -> Sign Message
4. Scan a QR code (e.g. "Hello world!")

Observe both errors in sequence.

For convenience, here's a `Hello world!` QR code:

[image]

# Requested changes
1. The first error be updated to mention Specter Desktop QR format
3. The second error message not occur

For the first change, we know that the user it trying to scan a message to sign, so we should be able to use that context to provide an improved error message.

# Root cause
I believe the root cause of the second error is that the view is trying to advance to a screen to show the user the message they are about to sign and that it should instead be returning to the main menu.

The first issue isn't really an error, but rather just an opportunity to improve usability.

# Code notes
I tried to dig into the code to figure out exactly what file and line I'm proposing be changed and I got stuck.

It looks like the Scan screen has no idea where it came from, which if that's true it means we can't provide that context-sensitive error message because we lack said context. I get this impressions from reviewing the code in ScanView::run(). If my interpretation is accurate, we can drop requested change 1 (improved error message). If it's possible to get that context, I maintain that a more specific error message that points the user in the right direction would be appropriate.

However that same function also show that the default next destination screen is NotYetImplementedView, whose class is implemented in view.py and its run method returns the MainMenuView, which seems like it should avoid that System Error screen... but it doesn't.

Taking a different approach, I decided to try to work my way backwards from SeedSignMessageConfirmMessageScreen. The stacktrace confirms that is called from SeedSignMessageConfirmMessageView. It looks like that is called from line 141, but for that to be reached, self.flow would have to have been set to FLOW__SIGN_MESSAGE in the SeedSelectSeedView constructor.

The only place where I see that occur is on line 1958, inside the SeedSignMessageStartView constructor.  OK, so how'd we get **here** when we say the first error try to send us back to the MainMenuView? Apparently from scan_views.py line 142!

But the only way we can get to line 142 is if the QR decoder's is_sign_message() method returns true, and that only happens if the message is a string that starts with "signmessage" (and our QR code did not).

...so I'm not sure where I went wrong here, but clearly I took a wrong turn at some point.

# Background
Issue https://github.com/SeedSigner/seedsigner/pull/262 added support for message signing. Users may expect this to allow them to scan a QR code with any message and SeedSigner will sign it. This is not true. The message needs to be wrapped in a the "Specter Desktop QR format". This format includes the derivation path as well as the message, and maintains compatibility with some other software called Specter..

This all is perfectly reasonable and I believe the formats should remain the same. If the documentation needs updated to explain how to use this feature, that will be a separate ticket.

jdlcdl, 2023-08-26:
Since you found this at the same commit as is our 0.7.0-rc1 release candidate, I will lean towards the simplest code fix which removes the second error.

Thank you for such a carefully written bug report.  I will try to provoke this error and may append to this comment below later.

hax0rbana-adam, 2023-08-27:
I can confirm that I'm seeing the same behavior as @jdlcdl. I only get the first error message (which, I agree, doesn't need changed for this release) when I go in via the scan menu.

kdmukai, 2023-08-28:
The bug is if you enter the message signing flow from the Seed Options menu.

Working on a fix.

PR #460 (2023-09-01): [bugfix] Reset state upon reading an invalid QR
fixes #456

## The problem
The BackStack and the `Controller.resume_main_flow` can create some confusing debugging when an error happens mid-flow.

In this case it was:
* Load a seed
* In Seed Options, select Sign Message
* Scan an invalid QR (not a format recognized by any of our decoders)

That would route you to the ErrorView which would then take you BACK, which routes you to SeedOptionsView. That detected the `Controller.resume_main_flow` which then routes to `SeedSignMessageConfirmMessageView` which assumes by that point that all necessary  [... cut]
commit 580cf3dc (2023-09-01): Reset state upon reading an invalid QR

fixes #456
closed 2023-09-01 by newtonick