SeedSigner/seedsigner #457: Bug Report: Enabling Persistent Settings w/ No MicroSD Inserted Produces Error

SeedSigner, 2023-08-27:
Steps to reproduce: Remove microSD card, scan any SettingsQR that includes enabling persistent settings.

Error produced includes text "persistent = 'E' is not valid"

Desired behavior: A more informative error message that either describes the specific issue in language more accessible to an end user, and/or directs the user to insert the microSD card if they wish to enable persistent settings.

kdmukai, 2023-08-29:
I would prefer to just silently ignore that setting in that scenario. Thoughts?

kdmukai, 2023-08-29:
Proposed fix via the "just ignore it" approach:

settings_views.SettingsIngestSettingsQRView:
```python
        self.config_name, settings_update_dict = Settings.parse_settingsqr(data)

        persistent_settings = settings_update_dict.get(SettingsConstants.SETTING__PERSISTENT_SETTINGS)
        if persistent_settings == SettingsConstants.OPTION__ENABLED and not MicroSD.get_instance().is_inserted:
            # SettingsQR wants to enable persistent settings, but no MicroSD is inserted.
            # For the sake of simplicity we just ignore that setting for now.
            # TODO: Can consider a warning screen instead that gives the user some options.
            del settings_update_dict[SettingsConstants.SETTING__PERSISTENT_SETTINGS]
            
        self.settings.update(settings_update_dict)
```

jdlcdl, 2023-08-29:
I support ignoring persistent settings for SettingsQR because it's appeal to me is to have custom settings for a device that remembers absolutely nothing when powered off.

If I only had a time machine,  I'd go back and argue that persistent settings, done in a manner where they could be enabled permanently so that seedsigner is writing to the microsd:
* just because I changed a setting or 
* just because I inserted my microsd that had a settings.json file, 
* or just because I scanned a SettingsQR with an extra E,
... instead of forcing me to knowingly take a deliberate action like "Save all settings to microsd now!", is too much "magic" (as I've opined in the past) for me.

My bias is based on the fact that I don't want anything being written to my microsd unless Im deliberately asking for that to happen.  I want to be able to dd my microsd frequently to get the hash of what's on it.  I understand that others will disagree.  This is just my opinion.

PR #464 (2023-09-03): [bugfix] Prevent SettingsQR from trying to enable Persistent Settings when SD card is not inserted
fixes #457 

## The problem
If you remove the SD card and then scan in a SettingsQR that attempts to enable Persistent Settings, you get a System Error exception.

## The fix
* Simply ignore the attempt to enable Persistent Settings when the SD card is not inserted.
* Refine the SettingsQR status message to reflect where the new settings are being stored:

[image] [image]

## Additional changes
* The test suite now mocks out the `MicroSD` class so that individual tests can run scenarios with vary [... cut]
closed 2023-09-03 by newtonick