SeedSigner/seedsigner #463: [0.7.0 rc1 Test]

jahangir13, 2023-08-28:
Testing the hosted settings-QR Generator:
'Message Signing' and 'Show QR brightness tips'  options in the Seedsigner Advance Settings menu are not included in the QR Generator:
[image]

kdmukai, 2023-08-31:
@jahangir13 thanks for catching this!

jdlcdl, 2023-09-01:
I see these two options have been added, 

and leave a reminder to @kdmukai that he noticed the SettingsQR Generator defaults camera rotation at 0 degrees while that changes to 180 degrees by default for 0.7.0.

kdmukai, 2023-09-01:
@jdlcdl the PR referenced above does include the new camera rotation default.

When I was testing yesterday I was using the public github.io version which, duh, is still pre-v0.7.0 until that update PR is merged.

closed 2023-09-03 by newtonick