SeedSigner/seedsigner #465: Seedsigner hangs when SD card is removed

deepfriedbrain, 2023-09-01:
Thanks for this wonderful project. 

I have built my seedsigner device and started to use it. As such it's working smoothly as expected, but I noticed that after successful boot if I remove the SD card, the device hangs and stops responding to the button inputs. I waited more than a minute to remove the SD card. I have persistent settings disabled.

Is this expected?

jdlcdl, 2023-09-01:
What you describe is not expected behavior.

Can you reproduce this over and over again?  It almost sounds like there is a problem with buttons, but then I'd expect that to occur with or without removing the sd card.

deepfriedbrain, 2023-09-01:
Yes, I can reproduce it consistently. Did it again after seeing your message, and it hanged within 5 secs of removing the card while I was just moving around on the 4 tiles of the home screen. With the SD card inserted, it has not hanged even once.

Reinserting the card doesn't do anything. The screen is completely frozen and would not respond. There's no green message either.

Note that I'm using it with RPi4 8GB temporarily (until I get a Zero), and it's using the only supported Waveshare hat.

Initially I thought it could be an issue with the power supply from my power bank. So, I connected with the official RPi 15W Power Supply, and could still reproduce the issue consistently.

deepfriedbrain, 2023-09-01:
Also, I'm not sure if it matters but thought I should mention it, the Seedsigner package was built on a RPi 3B manually from the source code following the instructions from the project docs. Now I'm using it with RPi 4. Does it matter?

jdlcdl, 2023-09-01:
I believe that the rpi 3b uses the same architecture as a the pi zero 2w.

For the rpi 4, you'd want to use the rpi4 image, but if an rpi3 was used to build for rpi4, then it might be fine.  Not sure.

The pre-release is here:
https://github.com/SeedSigner/seedsigner/releases/tag/0.7.0-rc1

The image you want is named seedsigner_os.dev.pi4.img and will have the following sha256 hash:
554756f0be105bfa184be2d9d1b67afc43e59b27bf359b59a8d035f65fd6b77c

kdmukai, 2023-09-01:
@deepfriedbrain Removing the SD card only works if you use a v0.6.0+ release image or build your own using SeedSigner OS: 
https://github.com/SeedSigner/seedsigner-os

It sounds like you followed the dev build installation steps in the main repo, which uses the original Raspberry Pi OS image. Can you confirm?

deepfriedbrain, 2023-09-01:
The version shown on the flash screen at startup is 0.7.0. I followed the instructions here:

https://github.com/SeedSigner/seedsigner/blob/dev/docs/manual_installation.md

The raspios image file I downloaded is named 2021-05-07-raspios-buster-armhf-lite.img

I did the installation around 2 weeks ago, but today was the first time actually using it. Should I do it again on the RPi4 itself?

I also noticed another case when it hangs intermittently even with the SD card in the slot. When I click on Scan and move the camera around, it hangs after about 10 secs. This is more intermittent. Happened twice today, but not reproducible right now.

jdlcdl, 2023-09-01:
LOL, I totally missed it.  Keith is spot on.  

You're running a Manual Installation which does not support removal of the microsd card.  It is indeed the latest version, but the way you're setup is more for development with a complete linux system.  The version which is meant to allow microsd card removal is the seedsigner-os build... a stripped down -- only what seedsigner needs linux system built specifically for seedsigner.  It has no usb data, no wifi drivers, no hdmi console.

If you want to develop your own seedsigner, or help seedsigner development, you've built the correct raspi-os version for development... but won't be able to remove your microsd card.

If you want to use your seedsigner in a much more secure manner, where you can remove the microsd card, you'll want to download the release candidate, or the official release which feels like it could be ready any day now.  And you can also build it yourself, reproducing the same binary as the official release.  For building that yourself, see https://github.com/SeedSigner/seedsigner-os 

Please excuse my confusion, I'd completely misunderstood this issue.

deepfriedbrain, 2023-09-01:
Thank you, it's clear now. It wasn't clear from the documentation. I'll build the seedsigner-os image this time and update the results.

I'll keep an eye on the pi4 freezing issue with the camera on. It happened with the official 15W power supply.

I truly appreciate the great support to go along with an awesome software.

kdmukai, 2023-09-01:
@deepfriedbrain for camera freezing issues, definitely try:
* different power supply (but a 9V battery connector will not work reliably)
* different usb cables
* double check both ends of the camera ribbon cable
* If using the standard Raspi camera (has its own daughterboard), the little camera module itself is just kind of stickered onto the daughterboard and that sometimes needs to be re-seated.

kdmukai, 2023-09-01:
@deepfriedbrain Oh, also consider updating the manual build instructions to make this distinction more clear (dev build uses Raspi OS to enable ssh, debugging output, etc BUT cannot remove the SD card vs SeedSigner OS for production purposes) in the docs. 

If you view the manual build instructions on github and just hit "." (period), that should launch github's web-based VS code environment which makes it easy-ish to edit and submit a PR.

deepfriedbrain, 2023-09-02:
Update - I was able to build seedsigner-os for rpi4. It took a good 5+ hrs on my RPi 400, but works flawlessly without the SD card now. Thank you so much!

closed 2023-09-02 by deepfriedbrain