SeedSigner/seedsigner #479: Add edit of BIP32 path on Xpub export

cryptochiggy, 2023-09-08:
It would be nice to be able to export a different BIP32 path besides the default m/48h/0h/0h/2h... maybe add an input of "account" or "index" like on the ColdCard UI, so you can move down the chain if desired. Like m/48h/0h/Xh/2h.

jdlcdl, 2023-09-08:
You can alter Advanced Settings: Script-type to include "Custom Derivation", and export an xpub for whatever path you like.  Can sign PSBTs like this too.

It's a bit hacky, in that it's not fully supported for things like "Address explorer" or "Address verification"

newtonick, 2023-09-15:
[image]

closed 2023-09-15 by newtonick