SeedSigner/seedsigner #480: `0.7.0` released resources do not contain version number

scgbckbone, 2023-09-12:
## 0.7.0

[image]

## 0.6.0

[image]

IMO it is better to keep version in resource filename for next release (as was in `0.6.0`)

jdlcdl, 2023-09-12:
I confirm!

This is the case in both versions of the release notes:
* single release version: https://github.com/SeedSigner/seedsigner/releases/tag/0.7.0
* multi releases version: https://github.com/SeedSigner/seedsigner/releases

newtonick, 2023-09-14:
This has been resolved. Thank you!

closed 2023-09-14 by newtonick