SeedSigner/seedsigner #483: UI issue when transcribing Seed phrase QR code template

Purchase73, 2023-09-14:
When backing up the seed by creating a SeedQR (Backup Seed, Export as SeedQR), and choosing Standard: 25 x 25, you get to the Transcribe SeedQR screen. In that screen, you go to each square (A1, A2, A3, etc), and mark on your paper template the dots that need to be filled. 
In 3 of the corners, the pre-printed alignment marker square sections show as squares. And the squares that you need to fill in are rounded, like you would make with a marker or sharpie pen.
However, in the bottom right pre-printed alignment marker square section, it shows up as dots. So you can't distinguish it from the rounded squares you're trying to fill in. 

I would expect the bottom right section to behave as the other 3 sections.

Purchase73, 2023-09-14:
https://t.me/c/1469623205/35339 
There's some discussion of the issue here

kdmukai, 2023-09-14:
Example:

[image]

The same dotted-but-should-be-squares rendering style appears in the zoomed in transcription UI.

PR #484 (2024-02-23): [Enhancement] Render SeedQR small registration block in solid squares
fixes #483

Renders the lower-right registration block in 25x25 and 29x29 SeedQRs as solid squares instead of dots.

Before vs after:
[image] [image]

And the same rendering improvement is continued in the zoomed-in transcription UI:
[image]

---

Supplements screenshot generator with:
* 12-word, standard SeedQR WholeQRView
* 24-word, compact SeedQR WholeQRView
* 24-word, standard SeedQR WholeQRView

--- 

Briefly tried and quickly gave up on getting the zoomed in transcription UI to render in the screenshot generator (just  [... cut]
commit 4d6e69ea (2024-02-23): [Enhancement] Render SeedQR small registration block in solid squares

fixes #483
closed 2024-02-23 by newtonick