SeedSigner/seedsigner #500: Suggest 12/24th words when entering a seed.

nostitos, 2023-10-10:
Currently we need to enter the seed twice to use a custom seed without a checksum word.

This could be included into the "enter 24-word seed" function.

It's already doing auto-correct with available words so it might as week suggest the only available option at that last step.

alorbach, 2025-03-14:
It would be great to be optional able to enter 12 word Seed without being forced to have used a 12th checksum word.
Normal users that want to use their own custom 12 word seeds crash into a wall with an error they don't understand ;).

Transisto, 2025-03-15:
> It would be great to be optional able to enter 12 word Seed without being forced to have used a 12th checksum word. Normal users that want to use their own custom 12 word seeds crash into a wall with an error they don't understand ;).

Are you aware the is a significant amount of entropy in the 12th word?  What you're suggesting is a bit like letting users use only 9 words if they feel like it and have SS assume the other words are `abandon` and the 12th checksum is the first possible word alphabetically.

alorbach, 2025-03-16:
To be honest I wasn't aware that BIP39 is only valid if the last word is calculated checksum.
Perhaps the  documentation could be enhanced for newbie users who are not aware of this fact.

notTanveer, 2025-08-15:
We can narrow down the valid final words, but the benefit varies significantly between 12-word and 24-word seeds due to their different checksum structures.

- 12-word mnemonic (128-bit entropy, 4-bit checksum):
With the first 11 words determined, approximately 128 out of 2,048 possible words will produce a valid checksum for the 12th position. This large number of valid options means narrowing it down doesn't provide much practical benefit.
- 24-word mnemonic (256-bit entropy, 8-bit checksum):
With the first 23 words known, only about 8 out of 2,048 possible words will satisfy the checksum requirement for the final position. This small set makes auto-suggestion highly valuable.