SeedSigner/seedsigner #502: Add seed generation via coin toss

nixonrichard1971, 2023-10-17:
A handy thing one has when helping out friends/family is a coin. It would be great if you could add the functionality of 12-word and 24-word seed generation with own entropy using coin toss. At the moment I need to carry an extra airgapped PC to calculate the 12th word and do a lot of manual calcs and check the list of BIP39 words. This could all be done within seedsigner, and that's the point of the dice roll, but just I don't carry 6 dices with me to visit my family / friends, but I carry the seedsigner in my pocket.

jdlcdl, 2023-10-17:
Thank you for bringing this up.

Coin toss entropy has been implemented in pr #442, but it is dependent on another to be merged first which is pr #404 (refactoring of the "mnemonic_generation.py" helper).  I'm going to assume that since you have a github account, that you are familiar with code, and so I politely request for your thoughts and feedback on pr #404, with hopes it gets merged sooner than later -- with ample time for peer review before a future release, so that coin toss entropy is available too.

Changes that need review are [here](https://github.com/SeedSigner/seedsigner/pull/404/files).  IMO: They're all safe, it's just that they're in a "delicate" section of code where peer review would be much appreciated.  Thanks in advance, if you can!

Everybody understands a coin toss, it's a great introduction to the topic of entropy, can easily be done by anyone under any circumstances (even if via multiple: "gotta go to the bathroom" excuses).

nixonrichard1971, 2023-11-03:
thanks for the reply. I can't really do a peer review of this, don't think I have the expertise sorry. maybe a wake up call to start getting familiar with coding bitcoin applications :)
Anyway I hope someone else could do it. I would really enjoy having the coin toss option!

jdlcdl, 2023-11-03:
> thanks for the reply. I can't really do a peer review...

No worries.  It's just my style to try to sucker new folks into peer review, because more eyes are better.

The most recent release was less than 2 months ago, and devs are taking a rest after the very busy integration and testing phase of that work.  As soon as development gets active again, everything is already in queue for this feature to progress forward.  Ill remember to ping you as it nears readiness, so that you can look forward to it in the following release (and to try to sucker you into reviewing the ui/ux or at least getting your thoughts on how it looks/works).

nixonrichard1971, 2023-11-04:
deal, thanks for the hard work!