SeedSigner/seedsigner #503: Failed to sign PBST

DFireTech, 2023-10-17:
Hello all!

I'm mobile currently, so, will add photos later on for context

# Issue
I import my seed to sign a pbst from BlueWallet to a hot wallet and when I get to the parts where I should be signing here is what happens:

Scan PBST from BW
Screen with "Select Seed" shows with `seed (?)`
Click through
"This seed does not add valid signature"

## Expected behavior
PBST is signed

# What I have done

I wonder if this is an issue I caused by having to swap Pi zeroes because my camera module stopped working on my other pi zero. 

Thanks for your time (and probable good criticisms)

DFT

jdlcdl, 2023-10-17:
I'm wondering if this is the same as a known issue where bluewallet had an xpub imported to create the wallet, but the format is not the same as seedsigner would export it.  Both types are valid, but if the wallet in bluewallet was created from an xpub exported directly from seedsigner, then it also has the wallet's master fingerprint... which helps seedsigner to identify which seeds are required to sign it.  If that's the case, then importing the xpub into bluewallet from seedsigner is the trick that works.  BUT, be sure that you have backups of your wallet just in case... before you go about deleting the wallet in bluewallet and trying to recreate it directly from seedsigner.

HoferBit, 2023-10-30:
> I'm wondering if this is the same as a known issue where bluewallet had an xpub imported to create the wallet, but the format is not the same as seedsigner would export it. Both types are valid, but if the wallet in bluewallet was created from an xpub exported directly from seedsigner, then it also has the wallet's master fingerprint... which helps seedsigner to identify which seeds are required to sign PSBTs. If that's the case, then importing the xpub into bluewallet from seedsigner is the trick that works. BUT, be sure that you have backups of your wallet just in case... before you go about deleting the wallet in bluewallet and trying to recreate it directly from seedsigner.

thank you, this is my issue

Admin661, 2024-05-30:
Is the issue resolved yet? You'll have to rectify with the BW weblink and try it again.

closed 2024-05-31 by newtonick

Radiokot, 2025-05-10:
@jdlcdl hey, thanks for the advice, this is still relevant for BlueWallet XPUBs.